ev3dev/ev3dev #1120: Brickman should display battery status in similar manor to EV3 software

lotu, 2018-06-26:
Brickman provides a battery status indicator that shows the current voltage of the battery.  This is useless to people who don't have a background in electronics and are aware of the details of the battery on the EV3.  Brickman is primarily used by teachers and students in elementary, middle, and high school this group is confused by seeing the voltage on the battery and does not help them in doing the single task it exists for; understanding if a devices batteries are full or not.

### What to do

The battery indicator should be replaced with one similar to what EV3 provides in the default software.  EV3's software defines a low and high voltage based on type of battery being used.

https://github.com/mindboards/ev3sources/blob/master/lms2012/lms2012/source/lms2012.h#L446

Then displays one of 5 pictures (empty battery to full battery) based on the current voltage using a simple linear interpolation where the low voltage is the empty battery and the high voltage is the full battery.
https://github.com/mindboards/ev3sources/blob/master/lms2012/c_ui/source/c_ui.c#L1660
https://github.com/mindboards/ev3sources/blob/master/lms2012/c_ui/source/c_ui.c#L1884

This solution is simple to implement and provides a useful, actionable, indicator to the user of the device.

### Related issues:
#68

ensonic, 2018-06-26:
@dlech if that sound good to you, I can give it a try.

dlech, 2018-06-26:
Yes, I was thinking about doing this as well. The tricky part is getting the values for full and empty since it depends on which type of battery is being used. In ev3dev-stretch, brickd knows these values. So, ideally, brickman would just get a percentage from brickd and use that to determine the fill area of the battery indicator.

Fabs37, 2022-01-02:
I think if you only have the accuracy of five different icons, you can ignore the difference between 6 AA batteries (9V) and the battery pack (10V). Since I guess that most of the ev3dev users are using the battery pack, I'm thinking of a grading like this:

```c
if       (voltage > 8.5) {
    icon = ICON_100
} else if(voltage > 7.5) {
    icon = ICON_75
} else if(voltage > 6.5) {
    icon = ICON_50
} else if(voltage > 5.5) {
    icon = ICON_25
} else {
    icon = ICON_0
}
```

Of course this is not exact, but way better than just a more or less meaningless number.

PS: The original firmware turns the status light to yellow when the battery is getting weak. I think this is a feature that could be implemented to ev3dev as well.