ev3dev/ev3dev #1121: apt-get update failed in image snapshot-ev3dev-stretch-ev3-generic-2018-06-13.img

gorozcoh, 2018-06-26:
apt-get update fails on image snapshot-ev3dev-stretch-ev3-generic-2018-06-13.img:

root@ev3dev-009:~# apt-get update
Err:1 http://security.debian.org stretch/updates InRelease
  Could not resolve 'security.debian.org'
Err:2 http://archive.ev3dev.org/debian stretch InRelease
  Could not resolve 'archive.ev3dev.org'
Err:3 http://httpredir.debian.org/debian stretch InRelease
  Could not resolve 'httpredir.debian.org'
Reading package lists... Done
W: Failed to fetch http://httpredir.debian.org/debian/dists/stretch/InRelease  Could not resolve 'httpredir.debian.org'
W: Failed to fetch http://security.debian.org/dists/stretch/updates/InRelease  Could not resolve 'security.debian.org'
W: Failed to fetch http://archive.ev3dev.org/debian/dists/stretch/InRelease  Could not resolve 'archive.ev3dev.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.

dlech, 2018-06-26:
Is you EV3 connected to the internet?

https://www.ev3dev.org/docs/networking/

dlech, 2018-06-26:
oh, actually, I think this is a duplicate of #1109.

dlech, 2018-06-26:
workaround: https://github.com/ev3dev/ev3dev/issues/1107#issuecomment-397082743

gorozcoh, 2018-06-27:
Thanks for your kind response. Why in this image imports fails? Example: from ev3dev2 import get_current_platform

ImportError: No module named 'ev3dev2'

dlech, 2018-06-27:
Because ev3dev-lang-python v2 hasn't been released yet.