ev3dev/ev3dev #1122: ev3dev not starting... also I dont appear screen "starting..." (mindstorms ev3)

inukisoft, 2018-06-27:
### What are you trying to do?

Install ev3dev for mindstorms. Im used etcher and last image of ev3dev. 

I installed sd (kingston 4g class 10 hc), orange led and wait 1 minute ... green led. Never appear splash screen, only "blank" screen".

### What did you expect to happen?

Splash screen "starting" for ev3dev and "starting" screen for ev3 mindstorm, but both not appear ... also dont appear update screen firmware.

### What actually happened?

not screen, not splash screen. For turn off I must take a battery and return. 

### What hardware and software are you using (including version numbers)?

brick hw v0.60
brck fw v1.10E
version of ev3dev: ev3dev-jessie-ev3-generic-2017-09-14

PD: It is possible that the screen is not appear because de aa batteries are used? (half life)

dlech, 2018-06-27:
Does the EV3 start without an SD card in it?

It could be low batteries.

inukisoft, 2018-06-28:
Plug my brick to step up regulator for 9volts, similar voltage that new batteries.... The principal problem is that not appear splash screen when I turn on. 

I believe that evdev charge from SD but the screen is off. Only appear main menu when boot ev3 software. 

I tryied change other firmware but without effect.

dlech, 2018-06-28:
If the LEDs went from blinking orange to solid green, then ev3dev booted. I'm not sure why you wouldn't see anything on the screen though.

BTTRG, 2023-09-03:
Old issue, but, I screwed up my install of ev3dev after installing a package that convinced dpkg it was a good idea to remove the ssh server. Reflashed my SD, now I get a blank screen where I should see the brickman splash. Solid green LED too. Gonna try reflashing again.