ev3dev/ev3dev #1123: Tacho Motor Encoder driver returns stale value at start

tcwan, 2018-06-30:
Currently the experimental PRU driver for capturing synchronous tacho motor encoder values returns [spurious / stale data](https://github.com/ev3dev/ev3dev/issues/853#issuecomment-401182903) on start. 

I'm wondering if there is some bug in the `pru_virtio_ring.h` `vring_init()` function:
```
#define vring_used_event(vr) ((vr)->avail->ring[(vr)->num])
#define vring_avail_event(vr) (*(__u16 *)&(vr)->used->ring[(vr)->num])

static inline void vring_init(struct vring *vr, uint32_t num, void *p,
                              uint64_t align)
{
	vr->num = num;
	vr->desc = p;
	vr->avail = (void *)((char *)p + num*sizeof(struct vring_desc));
	vr->used = (void *)(uintptr_t)(((uint64_t)&vr->avail->ring[num]
	                                + sizeof(uint16_t) + align-1) & ~(align - 1));
}
```

It could be due to `vring_used_event()` is referencing `vr->num` which is initialized to a non-zero value at the start.
However, the code is quite convoluted so I don't know for sure.

dlech, 2018-06-30:
As I [mentioned already](https://github.com/ev3dev/ev3dev/issues/853#issuecomment-401234968), the stale data values are a know bug in the iio framework and have nothing to do with the virtio rpmsg driver.

tcwan, 2018-07-01:
I was trying to understand why that is happening in the iio framework. It seemed like some kind of initial configuration bug in the framework, since by right if we do send messages from the PRU they are valid messages.

Subseqent `open` sessions do contain stale messages which were not flushed on the previous `close`. 

I believe that in any case, in terms of user mode API, we should have an option to flush the initial entries to make it less confusing to users.

dlech, 2018-07-01:
If you use libiio, it takes care of this for you (supposedly). However, I don't expect that we will be using these buffers in the final design anyway, so I don't think there is anything to be done here at the moment.

tcwan, 2018-07-01:
I saw that you reverted to using shared memory buffer for data transfer, so I guess it is a moot issue.

closed 2018-07-01 by tcwan