ev3dev/ev3dev #1125: make fails in ev3dev's tutorial example in  "Getting Started with C"

mardukeyo, 2018-07-08:
### What are you trying to do?

I am doing the "Getting Started with C" tutorial in ev3dev

I am using ev3dev-jessie-ev3-generic-2017-09-14 image and EV3 brick

There is a step where you have to install the compiler and some other tool features using the following commands

sudo apt-get update
sudo apt-get install build-essential

and I get the following error messages

===================================
robot@ev3dev:~/c-demo$ 
robot@ev3dev:~/c-demo$ sudo apt-get update
[sudo] password for robot: 
Err http://httpredir.debian.org jessie InRelease 
Err http://security.debian.org jessie/updates InRelease                  
Err http://archive.ev3dev.org jessie InRelease                            
Err http://httpredir.debian.org jessie Release.gpg                        
  Temporary failure resolving 'httpredir.debian.org'
Err http://security.debian.org jessie/updates Release.gpg                
  Temporary failure resolving 'security.debian.org'
Err http://archive.ev3dev.org jessie Release.gpg                         
  Temporary failure resolving 'archive.ev3dev.org'
Reading package lists... Done        
W: Failed to fetch http://httpredir.debian.org/debian/dists/jessie/InRelease  
W: Failed to fetch http://security.debian.org/dists/jessie/updates/InRelease  
W: Failed to fetch http://archive.ev3dev.org/debian/dists/jessie/InRelease  
W: Failed to fetch http://httpredir.debian.org/debian/dists/jessie/Release.gpg  Temporary failure resolving 'httpredir.debian.org'
W: Failed to fetch http://security.debian.org/dists/jessie/updates/Release.gpg  Temporary failure resolving 'security.debian.org'
W: Failed to fetch http://archive.ev3dev.org/debian/dists/jessie/Release.gpg  Temporary failure resolving 'archive.ev3dev.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
robot@ev3dev:~/c-demo$ 
===================================
robot@ev3dev:~/c-demo$ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bluez gnupg gpgv libbluetooth3
4 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,926 kB of archives.
After this operation, 220 kB disk space will be freed.
Do you want to continue? [Y/n] Y
WARNING: The following packages cannot be authenticated!
  libbluetooth3 gpgv gnupg bluez
Install these packages without verification? [y/N] y
Err http://security.debian.org/ jessie/updates/main libbluetooth3 armel 5.23-2+deb8u1
  Temporary failure resolving 'security.debian.org'
Err http://security.debian.org/ jessie/updates/main gpgv armel 1.4.18-7+deb8u4
  Temporary failure resolving 'security.debian.org'
Err http://security.debian.org/ jessie/updates/main gnupg armel 1.4.18-7+deb8u4
  Temporary failure resolving 'security.debian.org'
Err http://security.debian.org/ jessie/updates/main bluez armel 5.23-2+deb8u1
  Temporary failure resolving 'security.debian.org'
E: Failed to fetch http://security.debian.org/pool/updates/main/b/bluez/libbluetooth3_5.23-2+deb8u1_armel.deb  Temporary failure resolving 'security.debian.org'
E: Failed to fetch http://security.debian.org/pool/updates/main/g/gnupg/gpgv_1.4.18-7+deb8u4_armel.deb  Temporary failure resolving 'security.debian.org'
E: Failed to fetch http://security.debian.org/pool/updates/main/g/gnupg/gnupg_1.4.18-7+deb8u4_armel.deb  Temporary failure resolving 'security.debian.org'
E: Failed to fetch http://security.debian.org/pool/updates/main/b/bluez/bluez_5.23-2+deb8u1_armel.deb  Temporary failure resolving 'security.debian.org'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
robot@ev3dev:~/c-demo$ 
===================================

### What did you expect to happen?
I was hoping the compiler is already installed as part of the image on MicroSD and I still can compile the "Hello World" code using the make and makefile example given in the tutorial.

### What actually happened?
The "make' command fails as following

robot@ev3dev:~/c-demo$ make
-bash: make: command not found
robot@ev3dev:~/c-demo$ 

### What hardware and software are you using (including version numbers)?

robot@ev3dev:~/c-demo$ ev3dev-sysinfo -m

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

robot@ev3dev:~/c-demo$

BertLindeman, 2018-07-08:
Hi,
Quote from the tutorial:

> Before you start, make sure that you have configured a network connection to your ev3dev device and have opened an SSH connection to it.

DO you have a working internet connection on your EV3?
The messages on the ```apt-get update``` show there is no connection to e.g. '''http://httpredir.debian.org'''

Verify with ```sudo ifconfig``` and e.g. ```ping -c1 http://httpredir.debian.org```

tcwan, 2018-07-09:
You might want to take a look at this [guide](https://github.com/ev3dev/edu-docs/tree/master/docs). My version is slightly ahead of the ev3dev version due to fixes to URL links etc.

mardukeyo, 2018-07-10:
Hi Bert
Thank you for replying to my posting.
I followed the "Connecting to Internet via Bluetooth" step by step on Mac OS. The following snapshots shows the status of my setup.
However, when try ifconfig and ping test as you suggested,  I still think I am not getting through.
robot@ev3dev:~$ sudo ifconfigbnep0     Link encap:Ethernet  HWaddr cc:78:ab:50:20:74            inet addr:192.168.2.2  Bcast:192.168.2.255  Mask:255.255.255.0          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1          RX packets:1653 errors:0 dropped:247 overruns:0 frame:0          TX packets:850 errors:0 dropped:0 overruns:0 carrier:0          collisions:0 txqueuelen:1000           RX bytes:266204 (259.9 KiB)  TX bytes:97865 (95.5 KiB)
lo        Link encap:Local Loopback            inet addr:127.0.0.1  Mask:255.0.0.0          UP LOOPBACK RUNNING  MTU:65536  Metric:1          RX packets:243 errors:0 dropped:0 overruns:0 frame:0          TX packets:243 errors:0 dropped:0 overruns:0 carrier:0          collisions:0 txqueuelen:1           RX bytes:17628 (17.2 KiB)  TX bytes:17628 (17.2 KiB)
usb0      Link encap:Ethernet  HWaddr 02:16:53:56:4e:1e            UP BROADCAST MULTICAST DYNAMIC  MTU:1500  Metric:1          RX packets:0 errors:0 dropped:0 overruns:0 frame:0          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0          collisions:0 txqueuelen:1000           RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
usb1      Link encap:Ethernet  HWaddr 22:16:53:56:4e:1e            UP BROADCAST MULTICAST DYNAMIC  MTU:1500  Metric:1          RX packets:0 errors:0 dropped:0 overruns:0 frame:0          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0          collisions:0 txqueuelen:1000           RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
robot@ev3dev:~$ sudo ping -c 1 http://httppredir.debian.orgping: unknown host http://httppredir.debian.orgrobot@ev3dev:~$ 

Any idea?
Thanks againMarduke

  

The messages on the apt-get update show there is no connection to e.g. '''http://httpredir.debian.org'''Verify with sudo ifconfig and e.g. ping -c1 http://httpredir.debian.org—

BertLindeman, 2018-07-10:
Sorry, Marduke, MAC is not my cup of tea.

Some remarks trying to be helpful:

1. your latest post above is unformatted and thus hardly readible. 
That could be improved by putting three "back-tics" before the ifconfig output and also after that output.

2. you stated:
   > The following snapshots shows the status of my setup.

   Did you forget to add them?

Details on 1) to clarify:
Example of output UNformatted:

 ifconfig lo
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:358 errors:0 dropped:0 overruns:0 frame:0
          TX packets:358 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:39749 (39.7 KB)  TX bytes:39749 (39.7 KB)

And now with three "`" before and after:

```
 ifconfig lo
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:358 errors:0 dropped:0 overruns:0 frame:0
          TX packets:358 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:39749 (39.7 KB)  TX bytes:39749 (39.7 KB)
```

One thing to diagnose if your problem is related to the DNS (Domain Name Service).
I suggested `sudo ping -c1 httpredir.debian.org` **without `http://`**.
If this ping reports the host is unknown you may have a DNS problem.
Try to ping some dotted ip-addresses to verify that you do have an internet connection:
`sudo ping -c 1 8.8.8.8` 
This is a google public dns-address.
If this (or similar dotted address) works, there is a problem with your DNS configuration.
I do not have knowledge about network configuration on a MAC.

dlech, 2018-07-10:
FYI, markdown formatting and attachments are ignored when replying via email.

tcwan, 2018-07-11:
@mardukeyo 

I have not tried using Bluetooth Internet Sharing, I find that debugging Bluetooth issues is a pain. I'm connected using a USB cable.

I think you need to debug the network forwarding part first.
Since you are able to login to ev3dev, it means that your EV3<->PC connection is working.

Try the following:
```
robot@ev3dev:~$ host httpredir.debian.org
```
You should get something like the following:
```
httpredir.debian.org is an alias for static.debian.org.
static.debian.org has address 5.153.231.4
static.debian.org has address 128.31.0.62
static.debian.org has address 130.89.148.14
static.debian.org has address 149.20.4.15
static.debian.org has IPv6 address 2001:4f8:1:c::15
static.debian.org has IPv6 address 2001:67c:2564:a119::148:14
static.debian.org has IPv6 address 2001:41c8:1000:21::21:4
static.debian.org mail is handled by 0 .
```

If you get Error 53 (No DNS server), it means that somehow the network configuration is not setup properly. Normally this would be done using DHCP to define a nameserver but maybe the macOS Internet Sharing was not setup correctly for packet forwarding?

My nameserver config is setup automatically by ev3dev as follows:
```
robot@ev3dev:~$ cat /etc/resolv.conf 
# Generated by Connection Manager
nameserver 127.0.0.1
```
In addiiton, you should have a default route listed, and should be able to ping to the Gateway at least.

```
robot@ev3dev:~$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.2.1     0.0.0.0         UG    0      0        0 usb0
192.168.2.0     0.0.0.0         255.255.255.0   U     0      0        0 usb0
192.168.2.1     0.0.0.0         255.255.255.255 UH    0      0        0 usb0

robot@ev3dev:~$ ping 192.168.2.1 
PING 192.168.2.1 (192.168.2.1) 56(84) bytes of data.
64 bytes from 192.168.2.1: icmp_seq=1 ttl=64 time=1.50 ms
64 bytes from 192.168.2.1: icmp_seq=2 ttl=64 time=0.725 ms
^C
```

Then you can try pinging to outside network addresses like what @BertLindeman suggested. If that does not work, it means that packet forwarding on the macOS side is not working properly. In the Internet Sharing panel, you should allow the EV3 link to access Internet Sharing. If that still doesn't work I'm not sure what else is going on (other than antivirus or other network filtering going on in your Mac).

mardukeyo, 2018-07-11:
Hi TCWAN and Bert

Thank you all for all your help. Yes, it was the TCP/IP setting that was not being properly initialized for sharing internet through Bluetooth. Eventually,I was able to download the updates/compiler and compile/execute the "Hello World" code.

Best
Marduke

closed 2018-07-11 by mardukeyo

BertLindeman, 2018-07-11:
Thanks for reporting back.
Bert