ev3dev/ev3dev #1127: ev3 continually fails to install gcc-5/unstable due to slow internet connection

Jacob-Brink, 2018-07-11:
### I am trying to write c++ for ev3, but the ev3 cannot run the cross compiled programs because it can't install gcc-5 due to its slow connection wired through my computer. Sometimes it will take hours to find dependencies and then will auto abort due to a cut out in connection. Is there any workaround that does not involve having the ev3 os install anything? Thanks for all advice

dlech, 2018-07-12:
Use a cross-compiler that matches the GCC version on the EV3?

dlech, 2018-07-15:
https://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/

dlech, 2018-07-15:
By the way, is it just your EV3 that has a bad connection or do you just have bad Internet connection in general.

Also, if you don't fill out the questions in the template, it makes it harder to help you. For example, we have no idea which version of ev3dev you are using.

```
### What are you trying to do?

<!-- If you have a program that is not working, please post the *entire* program -->

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

<!-- Tip: copy and paste the output of `ev3dev-sysinfo -m` here -->
```

tcwan, 2018-07-16:
Also take a look at the guides [here](https://github.com/ev3dev/edu-docs).

dlech, 2018-07-17:
https://docs.docker.com/docker-for-windows/install/

Jacob-Brink, 2018-07-17:
oh, sorry about that