ev3dev/ev3dev #1129: import cv2 is extremely slow

AlexanderOMara, 2018-07-15:
Importing the `cv2` python module is extremely slow. In my testing it took `140` seconds to load, long enough that I initially assumed it would never complete. Is there something I can do to speed it up, so that it's practical to work with, or is the hardware simply not up to the task?

### What are you trying to do?

Python:

```py
import cv2
```

### What did you expect to happen?

Expected import to complete in a reasonably short period of time.

### What actually happened?

It takes a ***long*** time to import.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-07-08
Kernel version:     4.14.54-ev3dev-2.1.0-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653416CE4
BOARD_INFO_TYPE=main
```

dlech, 2018-07-15:
That sounds about right. I think I measured it around 2.5 minutes. On a Raspberry Pi 3, it still takes about 13 seconds.

Unfortunately, not much we can do about it besides just use OpenCV 2.

AlexanderOMara, 2018-07-15:
How much faster is OpenCV 2?

Also, I read that OpenCV 2 is Python 2 only, but ev3dev is Python 3 only? Is it still possible to use OpenCV 2?

dlech, 2018-07-15:
> How much faster is OpenCV 2?

dunno
 
> Also, I read that OpenCV 2 is Python 2 only, but ev3dev is Python 3 only? Is it still possible to use OpenCV 2?

https://github.com/ev3dev/ev3dev/issues/1038#issuecomment-365473646

dlech, 2018-07-15:
and if you don't want anything too fancy, you can always write your own python2 interface for motors and sensors - it is just reading and writing text to files.