ev3dev/ev3dev #1130: OUTPUT_A give serial0-0:MA not connected

adesito, 2018-07-17:
### What are you trying to do?
Hi i'm working in a project with mindstorm raspi and bickpi3 and I make lego roboarm set. With EV3 controller it work perfect, but when I migrate it to brickpi I've several issues related to the motor connections. Sensors works or at least I can configure them for example

Linux:
echo ev3-uart > /sys/class/lego-port/port1/mode
echo lego-ev3-color > /sys/class/lego-port/port1/set_device

Python code:
# import line
from ev3dev.brickpi import *

p = LegoPort(INPUT_2)
p.mode = "ev3_uart"
p.set_device = "lego-ev3-color"

if I do the same with motors i've the problem.

Linux:
echo "tacho-motor" > /sys/class/lego-port/port4/mode
echo "lego-ev3-m-motor" > /sys/class/lego-port/port4/set_device

echo: write error: Operation not supported
but a motorX Registered 'lego-nxt-motor' appears in the logs. I need medium motor, not large motor.

Python:
p = LegoPort(OUTPUT_A)
p.mode = "tacho_motor"
p.set_device = "lego-ev3-m-motor"

Exception: LegoPort(serial0-0:MA) is not connected

I don't know why serial0-0:MA is translated from OUTPUT_A. These motor  devices are in spi0.1:MA..B..C..not in serialX

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

 Linux version 4.14.54-ev3dev-2.2.0-rpi2 (david@freyr) (gcc version 6.4.1 20171012 (Linaro GCC 6.4-2017.11)) #1 SMP Sun Jul 15 15:06:58 CDT 2018
CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2

brickpi3 spi0.1: Address: 1
brickpi3 spi0.1: Mfg: Dexter Industries
brickpi3 spi0.1: Board: BrickPi3
brickpi3 spi0.1: HW: 3.2.1
brickpi3 spi0.1: FW: 1.4.6
lego-port port0: Registered 'spi0.1:S1' on 'spi0.1'.
lego-port port1: Registered 'spi0.1:S2' on 'spi0.1'.
lego-port port2: Registered 'spi0.1:S3' on 'spi0.1'.
lego-port port3: Registered 'spi0.1:S4' on 'spi0.1'.
lego-port port4: Registered 'spi0.1:MA' on 'spi0.1'.
lego-port port4: Added new device 'spi0.1:MA:lego-nxt-motor'
lego-port port5: Registered 'spi0.1:MB' on 'spi0.1'.
lego-port port5: Added new device 'spi0.1:MB:lego-nxt-motor'
lego-port port6: Registered 'spi0.1:MC' on 'spi0.1'.
lego-port port6: Added new device 'spi0.1:MC:lego-nxt-motor'
lego-port port7: Registered 'spi0.1:MD' on 'spi0.1'.
lego-port port7: Added new device 'spi0.1:MD:lego-nxt-motor'
tacho-motor motor0: Registered 'lego-nxt-motor' on 'spi0.1:MA'.
tacho-motor motor1: Registered 'lego-nxt-motor' on 'spi0.1:MB'.
tacho-motor motor2: Registered 'lego-nxt-motor' on 'spi0.1:MC'.
tacho-motor motor3: Registered 'lego-nxt-motor' on 'spi0.1:MD'.

Image file:         ev3dev-stretch-rpi2-generic-2018-06-13
Kernel version:     4.14.54-ev3dev-2.2.0-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000a2c7df57
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=BF9020BD5150324336202020FF140E30
BOARD_INFO_TYPE=aux

WasabiFan, 2018-07-17:
@adesito To answer your immediate question: `OUTPUT_A` is a constant defined by the ev3dev-lang-python library; it doesn't match what the motor driver on your device is called. Likely a recent change in the Stretch version of ev3dev, which is still in development and a bit unstable. You can replace `OUTPUT_A` with a string literal and use the value you're seeing and most things should work.

@dlech Are the port names from https://github.com/ev3dev/ev3dev-lang-python/pull/460 (released in https://github.com/ev3dev/ev3dev-lang-python/issues/459) still correct? What packaging was used there -- was that package released so it installs on Stretch but not Jessie? Is there anything else that has changed since which we need to release?

cc @dwalton76 @ddemidov

WasabiFan, 2018-07-17:
@dlech What am I missing?

https://github.com/ev3dev/ev3dev-lang-python/blob/stretch/ev3dev2/_platform/brickpi3.py

dlech, 2018-07-17:
Probably the top 6 commits from this tag: https://github.com/ev3dev/ev3dev-lang-python/commits/ev3dev-stretch/1.2.0

WasabiFan, 2018-07-17:
Ugh, those commits were made, tagged and released but never pushed to a release branch. So we're missing _some subset of_ those updates (some have been made semi-independently in `stretch`), but I don't know exactly what yet. Certainly the various changes to BrickPi support need to be re-done for our actual Stretch release branch now that the platform support structure has changed. The changelog will need to be reconciled and I'm not sure what we want to do with that release script -- I haven't been the one doing releases so that's up to whoever will in the future. I need to take a look at this later today and see what needs to be done.

Do we have a mechanism to leave the V1 series installed on Stretch in addition to V2? This is something we need to figure out soon as I hope to have everything ready for a Stretch release over the next few days now that the docs are merged.

dlech, 2018-07-17:
> Do we have a mechanism to leave the V1 series installed on Stretch in addition to V2?

Rename the debian package and make sure it doesn't try to install any files in the same place as the old package. (Doing the former should automatically take care of the latter.)

adesito, 2018-07-17:
thank you so much guys. I think I'm using "ev3dev.brickpi3" maybe I wasn't copy entire line, I'll review it.

The main issue, I guess, is with medium motor. I can't change de set_device to "lego-ev3-m-motor". I'll try to put the "spi0.1:MA" string instead of OUTPUT_A and try again.

Regards!

dlech, 2018-07-17:
BrickPi3 doesn't know the difference between a medium motor and a large motor (FWIW, also applies to BrickPi and PiStorms).

Why do you need to set it to medium motor? Is using the default NXT motor good enough?

adesito, 2018-07-18:
Hi dlech I'll try large motor but I think they have different rotation specs or something like that, but as I write I'll try with largemotor instead of medium.

On the other hand, I'm putting the string directly at objet creation as p = LegoPort('spi0.1:S4') for example. But the result for this sensor, when I change the mode to TOUCH, is device not connected... and the sensor is created in brickman and I can see it and manage it.

CODE Python:
p = LegoPort('spi0.1:S4')
p.mode = "ev3-analog"
p.set_device = "lego-ev3-touch"
self.base_limit_sensor = TouchSensor('spi0.1:S4')
self.base_limit_sensor.mode = self.base_limit_sensor.MODE_TOUCH

LOG:

2018-07-18 06:46:33,077     INFO: TouchSensor(spi0.1:S4): path None, attribute mode
2018-07-18 06:46:33,078 CRITICAL: TouchSensor not present in port S4 - TouchSensor(spi0.1:S4) is not connected
2018-07-18 06:46:34,081    DEBUG: Using selector: EpollSelector

Regards!

dlech, 2018-07-18:
You probably need to add a short delay after `set_device`, e.g. `time.sleep(0.5)`.

dlech, 2018-07-18:
> Pycharm warned me that brickpi3 is not a valid reference :( is this correct? maybe have I a deprecated version of python in this ev3dev linux image?

You probably have an old version in PyCharm, and a newer version on the EV3.

dlech, 2018-07-19:
@ev3dev/python, can we get a v1.2.0 release on PyPi?

WasabiFan, 2018-07-19:
Travis CI is configured to do it when it runs on a tagged branch; its config hasn't been updated for the new repo location though. Given that this is a tag rather than a branch it won't be quite as easy as just pushing a change to that YAML property, but I can probably trigger a build via the website. I'll check in a bit. If that doesn't work I think @ddemidov will have to do it manually.

dlech, 2018-07-19:
Ah, that explains why we don't have any other debian package tags in that repo (the tag that includes the `ev3dev-stretch` prefix).

I have a hunch that after you fix the repo name, it will work. And I think it might be a good idea to add a conditional so that debian tags don't get deployed to PyPi, e.g if the tag contains a `/`, don't deploy it to PyPi.

https://docs.travis-ci.com/user/deployment#Conditional-Releases-with-on%3A

dlech, 2018-07-19:
And you will need to create a new tag that is just `1.2.0`.

WasabiFan, 2018-07-19:
I don't get it... why do we need tags specific to the Debian package? Why not deploy to all release channels from the same tag?

WasabiFan, 2018-07-19:
Something like this?

```diff
diff --git a/.travis.yml b/.travis.yml
index 1be1447..06dcf96 100644
--- a/.travis.yml
+++ b/.travis.yml
@@ -13,4 +13,5 @@ deploy:
     secure: W13pXc2pp2A9gBUDz2QW/K3OfDT/cn/iapkW6NMdkwRQ+CL7MWXsY2qayfxs6QPfR5W7838pmMXogLOie70n8Hov5/XIrou+dEz/xh6If2VFG41KUz/tP9cy1yUhDYeADn5I+mr+8Yrh59OkHkaeZH1EFndbsxY4pyXV0DT+FNVg4eLc8sE5ZCfa/itVdsqp8M4xrMe8NO/ldnFEIHFymyEMTZHR5qoJD+Uk+AGMR4cSeSuOrZixnSLpQViZDFHpwej6F1LrreLltbT9ChjSEmdPAr1Jp1ReQfuD+vBzUMkhVfZyEo+fY1x8FZVPrTAEtbJGDhvTAsrV1KTgPqanMyyrCIG+OWfCSYYPyg7MYbAJVfB+P0BRp1cm2D5oFrpAZE129oVtATOykQarFzzRFhH4Tyc04WriyY/greEAe58MqYoJAZXUIe/JLf9+GLdjBKD07+q7QMZvyEEdsDCPYkqKQeuwrWZ/JlpWO5rmy12L23aYzvJqhcKo9LZQOY8LkmFmxuxt7k5eb/3iZ1trhj/lwoLLSu8l29B8cK3dax38URie0x9bMMhQRxaO59JQuhGuu0sNHiuFQHs6wLs/V8ff1IZIlRSlzztqIP3KW514TEdsFIuF0Gpn5wvagAXFbSnyxAUUTy81eQjY/ExTiUHKfU3zjluaAYGt9zjF0Bc=
   on:
     tags: true
-    repo: rhempel/ev3dev-lang-python
+    condition: $TRAVIS_TAG != ev3dev-stretch/*
+    repo: ev3dev/ev3dev-lang-python
```

You had mentioned elsewhere that it would be good to maintain a branch for the 1.x series. So how about I push that (which is based on the `ev3dev-stretch/1.2.0` tagged commit) to `stretch-1.x`?

dlech, 2018-07-19:
👍 Looks good to me.

WasabiFan, 2018-07-19:
Minor snag: PyPi is unhappy with @ddemidov's account settings 😆:

https://travis-ci.org/ev3dev/ev3dev-lang-python/builds/405654908#L633

> HTTPError: 400 Client Error: User 'Denis.Demidov' does not have a verified primary email address. Please add a verified primary email before attempting to upload to PyPI. See https://pypi.org/help/#verified-email for more information.for more information. for url: https://upload.pypi.org/legacy/

@ddemidov can you verify your email address? Once that's done we can re-run the Travis build and it should upload successfully.

ddemidov, 2018-07-19:
> can you verify your email address? 

done

ddemidov, 2018-07-19:
I had to reset my forgotten password in the process, and now I think I need to reencrypt it for travis.

ddemidov, 2018-07-19:
Also, we probably want to cherry-pick the travis changes into other branches

WasabiFan, 2018-07-19:
@ddemidov 

> Should I force-update the 1.2.0 tag?

Yeah, I think that's fine. Make sure it's an annotated tag so the build will pick it up. I'll have to figure out how to reconcile a force-pushed tag on my end but I doubt anyone else has pulled.

> Also, we probably want to cherry-pick the travis changes into other branches

Agreed. Once it works in this case I'll open a PR to cherry-pick.

ddemidov, 2018-07-19:
> I'll have to figure out how to reconcile a force-pushed tag on my end but I doubt anyone else has pulled.

Something like `git tag -d 1.2.0` followed by `git fetch --tags` should work

ddemidov, 2018-07-19:
For some reason Travis does not consider this a tagged commit. Do you have any ideas how to change its mind?

WasabiFan, 2018-07-19:
You need the `-a` flag IIRC; it must be an "annotated" commit, for reasons I couldn't explain. You can check what annotstion message I used and copy that.

WasabiFan, 2018-07-19:
Also, `git push --tags` rather than `fetch`

ddemidov, 2018-07-19:
Yes, I did exactly that.

ddemidov, 2018-07-19:
May be I'll try to drop the last commit and the repush it.

ddemidov, 2018-07-19:
Fetch was meant for you to update the tags locally :)

ddemidov, 2018-07-19:
The tag [is ok](https://github.com/ev3dev/ev3dev-lang-python/tree/1.2.0). May be travis has old 1.2.0 cached somewhere.

dlech, 2018-07-19:
I just deleted and and re-pushed the `1.2.0` tag to GitHub and Travis seems to have picked it up.

dlech, 2018-07-19:
@adesito, you should be able to upgrade the python-ev3dev version in PyCharm now (to 1.2.0) and get the `import brickpi3` working.

WasabiFan, 2018-07-19:
Can confirm that Travis succeeded! :tada:

[image]

adesito, 2018-07-23:
thank you so much guys

Regards!

closed 2018-07-24 by dlech