ev3dev/ev3dev #1132: run-direct not active until sending nonzero duty value

laurensvalk, 2018-07-18:
When setting a motor to run-direct mode, even when explicitly writing the duty value 0, the mode does not become active. It activates as soon as you write a nonzero value. Writing a 0 afterwards does have the expected effect though.

Expected: Writing 0 should always have the same effect

To reproduce:

```
# Boot EV3 with a motor attached (mine is a large motor on port D)
cd /sys/class/tacho-motor/motor0/
echo run-direct > command
# Try to rotate motor by hand: easy --> duty mode not yet active
echo 0 > duty_cycle_sp
# Try to rotate motor by hand: easy --> duty mode still not active
echo 1 > duty_cycle_sp
echo 0 > duty_cycle_sp
# Try to rotate motor by hand: hard --> duty mode now active (expected behavior)

```

```
Image file:         ev3dev-stretch-ev3-generic-2017-12-13
Kernel version:     4.9.58-ev3dev-1.6.0-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016533DDB29
BOARD_INFO_TYPE=main
```

dlech, 2018-07-18:
I suspect this is still a problem with the `ev3dev-2.2.0` kernel, but you are using a rather old kernel.

dlech, 2018-07-18:
You probably need to run `sudo flash-kernel` manually. see #1110

dlech, 2018-07-26:
Fix released in `ev3dev-2.2.1` kernel.

closed 2018-07-26 by dlech