ev3dev/ev3dev #1133: Battery Voltage reading inconsistent in stretch

tcwan, 2018-07-19:
### What are you trying to do?

View the battery voltage and current values in Brickman, on ev3dev Jessie (release version) and Stretch (2018-7-15 snapshot).

### What did you expect to happen?

Readings to be similar.

### What actually happened?

On Jessie: 8.98 - 9.00 V, 163 - 169 mA
On Stretch: 9.13-9.14V, 149 - 156 mA

### What hardware and software are you using (including version numbers)?

(Only stretch version shown)

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-07-15
Kernel version:     4.14.55-ev3dev-2.2.0-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535BF127
BOARD_INFO_TYPE=main
```

tcwan, 2018-07-19:
I'm using an external switching power supply.
According to my DVM, it should be close to 9.00V, unloaded. But it can fluctuate a bit.
On jessie, I've seen Brickman report values from about 8.95V to 9.05V.

tcwan, 2018-07-19:
In the `jessie` driver, I see:
```
	return legoev3_analog_batt_volt_value(bat->alg) * 2000
		 + legoev3_analog_batt_curr_value(bat->alg) * 1000 / 15 + 50000;
}
```
In the `stretch` driver, I see:
```
		val->intval += 200000;
```

The constant added to `stretch` driver is 200,000 vs. 50,000 for `jessie`. The rest of the formula seems to be identical.

dlech, 2018-07-19:
Hmm... I'm not sure why I would have changed that value in stretch. It sounds like jessie is the more accurate one.

The lower current draw in stretch is interesting though since it doesn't depend on the voltage calculation. Did you have any devices (motors, sensors, USB) attached to the EV3 when you recorded these values? Were the LEDs in the same state?

tcwan, 2018-07-19:
It was just in the idle state after bootup. I think the LEDs are lighted similarly in both jessie and stretch, right?

Both images were booted on the same hardware configuration, so no change there.
I have two Large and one Medium motor attached (but idle), and one IR sensor and one Touch sensor connected.

dlech, 2018-07-19:
Looking at my EV3 right now, with `ev3dev-2.2.0` kernel (plus fix for #1131 - although it shouldn't make a difference), voltmeter is showing 9.02V and EV3 is showing 9.05V. The EV3 value changes a few hundredths of a volt when motors and sensors are added or removed while the voltmeter doesn't really change.

Maybe I'll try on another EV3 later or we can wait and see if the next kernel release makes a difference for you.

Also, if you can test the voltage while the supply is actually connected to the EV3, we might find that there isn't a discrepancy after all.

tcwan, 2018-07-20:
With the switching power supply connected to the idle EV3:

Jessie: 8.96-8.98V on Brickman, 9.00V on DVM
Stretch (2018-07-15 snapshot): 8.94-9.14V on Brickman, 9.00V on DVM
It started off at the high side > 9.1V and then slowly drifted downwards. However, the readings does not seem to be as stable as Jessie. Not sure why it won't give a consistent reading compared to the last time.

dlech, 2018-07-20:
I did some more testing yesterday and found that the current reading seems to be quite accurate (although precision is only 6.5mA). The voltage, on the other hand, tends to drop at the current goes up.

Large motor running at 100% duty cycle:

| Condition | Brick Current | Brick Voltage | External Voltage |
|-|-|-|-|
| No load | 215mA | 9.05V | 9.02V |
| Load | 560mA | 8.86V | 9.01V |
| Locked rotor | 867mA | 8.72V | 8.99V |

So, it seems there is some kind of voltage drop in addition to the shunt resistor that isn't being accounted for.

Do you see the same thing if you run a motor? (My load is not precise - I just slowed the motor down with my hand, but you could at least test the no load and locked rotor cases.)

dlech, 2018-07-20:
P.S. It would be helpful for anyone to do this test, not just @tcwan. :wink:

dlech, 2018-07-20:
I found my voltage drop. It was in the wiring between the power supply and the EV3. After connecting the volt meter directly to the EV3, I am seeing:

| Condition | Brickman Current (mA) | Brickman Voltage (V) | Voltmeter (V) |
|-|-|-|-|
| No load | 221 | 9.07 | 8.93 |
| Locked Rotor | 879 | 8.78 | 8.64 |

This matches up with the 0.15V expected discrepancy.

dlech, 2018-07-20:
https://github.com/ev3dev/ev3-kernel/commit/12ffcae55050d9627f42b3525465e142d72a597e

tcwan, 2018-07-21:
I took the inline measurements by probing the power supply plug connector. I think that may have stretched the wire and changed the contact resistance for the EV3 battery terminals, leading to the lowered voltages read by Brickman.

dlech, 2018-07-26:
Fixed released in kernel `4.14.58-ev3dev-2.2.1-ev3`.

closed 2018-07-26 by dlech