ev3dev/ev3dev #1135: Wi-Fi fails to connect with stretch on FATCATLAB device

kortschak, 2018-07-21:
### What are you trying to do?

Connect to device via Wi-Fi.

### What did you expect to happen?

Happy connection.

### What actually happened?

Failure to connect.

After scanning fine and showing services and then performing a connect, connmanctl gives me "Agent request cancelled by ConnMan" and "Error /net/connman/service/wifi_<stuff>_managed_psk: Operation aborted", and dmesg shows:

```
[  290.537615] R8188EU: indicate disassoc
```

Wi-Fi is fine on jessie for this device and the relevant module is loaded in stretch.
```
robot@fatcat:~$ lsmod | grep 8188
r8188eu               454656  0
cfg80211              622592  1 r8188eu
```

Also note that `lsusb` does find the device and scanning is successful.

### What hardware and software are you using (including version numbers)?

```
robot@fatcat:~$ ev3dev-sysinfo 
Image file:         ev3dev-stretch-bone-generic-2017-09-30
Kernel version:     4.14.54-ev3dev-2.2.0-bb.org
Brickman:           0.10.0
BogoMIPS:           995.32
Board:              board0
BOARD_INFO_HW_REV=00C0
BOARD_INFO_MODEL=A335BNLT
BOARD_INFO_SERIAL_NUM=1116BBBK1052
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_MODEL=FatcatLab EVB
BOARD_INFO_TYPE=aux
```

kortschak, 2018-07-21:
It's a TPLink 725.
```
robot@fatcat:~$ lsusb 
Bus 001 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
```

From dmesg:
```
[    2.751459] usb 1-1: new high-speed USB device number 2 using musb-hdrc
[    2.903847] usb 1-1: New USB device found, idVendor=0bda, idProduct=8179
[    2.910793] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.918115] usb 1-1: Product: 802.11n NIC
[    2.922287] usb 1-1: Manufacturer: Realtek
```

dlech, 2018-07-21:
apparently people like to practice sending code style patches on this driver: https://gist.github.com/dlech/b122356b3332292bc44d1d3077622595

don't make it easy to see what has actually change :disappointed: 

I don't have one of these, so I can't be of much help. The only think I know to suggest is to play around with the [module parameters](https://github.com/ev3dev/bb.org-kernel/blob/ev3dev-stretch/drivers/staging/rtl8188eu/os_dep/os_intfs.c#L102). I know turning off power management seems to help on some other RealTek devices.

Beyond that, you would have to do a git bisect to see what actually broke it.

dlech, 2018-07-21:
Maybe this will help?

https://github.com/torvalds/linux/commit/4004a9870bbefdb6644c3d2033f5315920a3b669#diff-b5ed99b43c60c6b8e71e5ed08d9ee62a

dlech, 2018-07-21:
nevermind, that is already applied

kortschak, 2018-07-21:
That patch is horrific. I'm in the middle of teaching at the moment, so I won't be able to get to this any time soon, but I will look into it when I can. I also need to check whether it work on the EV3 (I think it does).

dlech, 2018-07-21:
Come to think of it, I think we may have an alternate out-of-kernel driver on EV3 for this dongle (which would explain the difference if it does indeed work on EV3).

kortschak, 2018-07-21:
Confirmed working on the EV3.