ev3dev/ev3dev #1136: Cross compiling for stretch?

obo, 2018-07-26:
I have never cross-compiled anything, but I would like to get a rather big library (aruco) compiled on EV3 running ev3dev-stretch. It takes ages and often fails probably due to system overload.

I found the guide https://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/

Is it OK to use the only available https://hub.docker.com/r/ev3dev/debian-jessie-cross/ for stretch? (I also have no real experience with docker so far.)

tcwan, 2018-07-26:
No, if you're using jessie Docker image it'll be linked against old libraries which may not be compatible with the stretch libraries. 

Take a look [here](https://github.com/ev3dev/edu-docs) for some info regarding setting up a cross-compilation environment for ev3dev. There is no TOC currently, so you need to poke around the different subdirectories in doc.

I have a customized version of the stretch cross compiler tools available from [here](https://hub.docker.com/r/tcwan/debian-stretch-cross-pytools/) which has been tailored for use with [ev3dev-c](https://github.com/in4lio/ev3dev-c).

dlech, 2018-07-26:
Also, FYI, we have backported OpenCV 3 to ev3dev-stretch, but we have since found it is so slow, it is hardly usable (e.g. #1129).