ev3dev/ev3dev #1138: Brickpi3 not detected by ev3dev stretch

mingan666, 2018-07-30:
### What are you trying to do?

Trying to startup my ev3dev stretch together with the brickpi3 and rasbian pi 3 model B+

I've tried to update the firmware through rasbian for robots, but that did not help. Previously I was able to get at least a sensor to work with python on rasbian for robots, but I want to be able to use java instead. That's why I want to get ev3dev working.

### What did you expect to happen?

Output from dmesg to show brickpi info. Also files under /sys/classes.

### What actually happened?

Output from dmseg:
```
robot@ev3dev:/var/log$ dmesg | grep brick
robot@ev3dev:/var/log$ 
```

And contents of for example power_supply:
```
robot@ev3dev:/sys/class/power_supply$ ls -al
total 0
drwxr-xr-x  2 root root 0 Jul 29 20:16 .
drwxr-xr-x 54 root root 0 Nov  3  2016 ..
robot@ev3dev:/sys/class/power_supply$ 
```

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2018-07-26
Kernel version:     4.14.56-ev3dev-2.2.1-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Board:              board0
BOARD_INFO_HW_REV=A020D3
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Plus Rev 1.3
BOARD_INFO_SERIAL_NUM=000000003e6b500d
BOARD_INFO_TYPE=main
```

This information is taken when running rasbian for robots troubleshooting script. As it shows the brickpi is working fine there. 

**BrickPi3 Troubleshooting Script log**
```
Checking for hardware, and checking hardware and firmware version.
==================================================================
Manufacturer    :  Dexter Industries
Board           :  BrickPi3
Serial Number   :  F247E348504D5741372E3120FF0E1F11
Hardware version:  3.2.1
Firmware version:  1.4.6
Battery voltage :  4.211
9v voltage      :  9.132
5v voltage      :  4.849
3.3v voltage    :  3.334
```

Config file contents from /boot/flash/config.txt can be found here: https://pastebin.com/sChbSduz

WasabiFan, 2018-07-30:
Did you uncomment the line in `config.txt` that says

> uncomment if you are using BrickPi3

?

mingan666, 2018-07-30:
I'm so sorry... I was sure I had done that... 
But when I double checked it seems like I never saved the file.
It seems to be starting the way it should now.

closed 2018-07-30 by mingan666

WasabiFan, 2018-07-30:
:laughing: no problem — I was quite confused by the fact that you had attached the file, indicating that you knew you had to edit it, but it wasn't edited!