ev3dev/ev3dev #1139: Connecting ps4 controller cause ev3 to lag

hbc2, 2018-07-31:
System information:
I am using: LEGO MINDSTORMS EV3
I installed ev3dev using this image file: ev3dev-jessie-ev3-generic-2017-09-14.zip
My kernel version is: v4.4.87-22-ev3dev-ev3
My host computer is running: Win10
I am connecting using: USB for ssh
I am connecting and pairing over Bluetooth with the ps4 controller

About my issue:
Hi,

As near as I can tell, I'm having the same issue as in #707 which is closed 
 
When I pair the ps4 to the ev3, there is a serious lag. Using top I see the brickman process goes to 40% and khidpd goes to 20%.  The lag pretty much makes using the ev3 impossible.

I tried to follow what you all were discussing in #707 but, I'm just not at the level you're all at so I don't know how to troubleshoot further. 

Thanks in advance.

dlech, 2018-07-31:
The patch is still in the ev3dev-jessie kernel: https://github.com/ev3dev/ev3-kernel/commit/e247391c9df47e3c7c9414f2905b50f133e94f54

0xB0 is also the slowest value listed in the comments: https://github.com/ev3dev/ev3-kernel/blob/0d17bab6c1fe53a9ce0d8ba6b3d89c6b35a56ca0/drivers/hid/hid-sony.c#L1909

So, I think without the patch, the situation would be even worse.

On ev3dev-stretch, it looks like there should be a `bt_poll_interval` attribute in sysfs that you can play around with. If you find out what the slowest rate is, then we can make that the default.

hbc2, 2018-08-04:
Thanks, delech.

I flashed snapshot-ev3dev-stretch-ev3-generic-2018-07-26.img.xz, paired the ps4 controller and could see immediately the lag issue has been resolved.

The bt_poll_interval value is set to 4 in the 2018-07-26 image.  With that setting I noticed the device would periodically miss some values.  Frequently the end point values (0 and 255) for some of the physical controls wouldn't be reported.  I upped the poll value and found 8 to be the best for me.  In this way I can now stop the motor when 0 is reported and be sure the motor stops.  

Thanks again.

hbc2, 2018-08-04:
btw, from your comment, I was unclear if "the ev3dev-jessie kernel" was in the  snapshot-ev3dev-stretch-ev3-generic-2018-07-26.img.xz  image.  I can confirm, though, with that image I don't get the lag.

dlech, 2018-08-04:
There were a lot of changes to the sony driver in the mainline kernel between ev3dev-jessie and ev3dev-stretch, so it is hard to say what has changed.

What is the CPU usage with `bt_poll_interval` set to 8? We can look at making this the default value, or perhaps even a little higher.

dlech, 2018-08-06:
FYI, I've made 10ms the default time in the `ev3dev-2.2.2` kernel. We can re-open this issue if we need to change it again.

dlech, 2018-08-06:
https://github.com/ev3dev/ev3-kernel/commit/fe029e4f0e9523221db43c33d01c0962ae2ed44d

closed 2018-08-06 by dlech

hbc2, 2018-08-10:
Sorry for the delay in getting back.  Below is a table for the cpu usage I saw.  (note: with val = 4, packets were lost due to some lag)

val min:CPU max:CPU
4 min:9.5 max:11.7
8 min:4.9 max:6.2
10 min:3.9 max:5.0

An interval of 10 seemed fine and (obviously) gave the best usage.  I will give ev3dev-2.2.2 a try.

Thanks.