ev3dev/ev3dev #1140: Sometimes not holding status received on tacho motors

adesito, 2018-08-03:
### What are you trying to do?
I'm programming a lego roboarm (standard from lego mindstorm). Roboarm receive commands from web server (tornado 5.1 version non blocking httpserver for long polling and long time request).

I send commands to the robot like initialize, start movement, stop, get temperature...

### What did you expect to happen?
when motor reach to a rel or abs position I expect to get a "holding" status in the motor. Example
self.base_motor.run_to_rel_pos(speed_sp=SPEED_BASE, position_sp=direction * -self.base_position)
while self.base_motor.STATE_HOLDING not in self.base_motor.state \
                    and self.base_motor.STATE_OVERLOADED not in self.base_motor.state:
           pass

motors stop action is set to stop_hold

### What actually happened?
sometimes motors remains in while loops indefinitely beacuse not holding or overloaded received, only running...

Example Log OK status:

2018-08-03 18:09:59,864    DEBUG: [MOVE][BASE-MOTOR] MOVE_1 value: 1 Status: ['running'] shutdown_flag: False
2018-08-03 18:09:59,867    DEBUG: [MOVE][BASE-MOTOR] MOVE_1 value: 1 Status: ['running'] shutdown_flag: False
2018-08-03 18:09:59,895    DEBUG: [MOVE][BASE-MOTOR] MOVE_1 value: 2 Status: ['running', 'holding'] shutdown_flag: False

Example Log KO status no holding (motor reach the position but nothing happens):

2018-08-03 18:09:39,812    DEBUG: [MOVE][BASE-MOTOR] MOVE_1 value: 301 Status: ['running'] shutdown_flag: False
2018-08-03 18:09:39,815    DEBUG: [MOVE][BASE-MOTOR] MOVE_1 value: 301 Status: ['running'] shutdown_flag: False
2018-08-03 18:09:39,817    DEBUG: [MOVE][BASE-MOTOR] MOVE_1 value: 301 Status: ['running'] shutdown_flag: False
2018-08-03 18:09:39,820    DEBUG: [MOVE][BASE-MOTOR] MOVE_1 value: 301 Status: ['running'] shutdown_flag: False
2018-08-03 18:09:39,822    DEBUG: [MOVE][BASE-MOTOR] MOVE_1 value: 301 Status: ['running'] shutdown_flag: False

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2018-07-15
Kernel version:     4.14.56-ev3dev-2.2.1-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000a2c7df57
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=BF9020BD5150324336202020FF140E30
BOARD_INFO_TYPE=aux
```
I'm using ev3-python 1.2.0 (last version you released a week or two ago, brickpi3 push ;))

Regards

dlech, 2018-08-06:
I've made a change that might possibly fix this in kernel `ev3dev 2.2.2`. Please test.

adesito, 2018-08-06:
Thank you so much. I'll test tomorrow.

I create a timeout too to avoid thid kind of problems, but I think oits
better if the motor respond correctly.

Than you so much

Best Regards

adesito, 2018-08-08:
Hi guys I tested it and I think is working better than before, now the robot works and detects the holdings....I launch the movement for 5 minutes or more and no holding missing issues.

I don't know if I can ask here but is there any way of detect that a sensor in the brickpi3 I was disconnected??

Regards!

dlech, 2018-08-08:
Closing since this seems to be fixed.

The BrickPi3 firmware doesn't have the ability to detect when sensors are attached or removed. Feel free to open a new issue for any other questions.

closed 2018-08-08 by dlech