ev3dev/ev3dev #1141: root password maker does not work

BothellRobotics, 2018-08-04:
### What are you trying to do?
I am trying to SSH EV3 after booting with EV3DEV.  Once I enter ssh root@<IP address of EV3 brick>, I have been prompted password. When I enter "maker", I get  permission denied. 

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

BertLindeman, 2018-08-04:
Think you are missing the letter b in the userid. `maker` is the password set for user `robot`
See [ev3dev doc connecting-to-ev3dev-with-ssh](https://www.ev3dev.org/docs/tutorials/connecting-to-ev3dev-with-ssh/)

BothellRobotics, 2018-08-04:
Thank you. You saved my day. For user name robot the password works.

closed 2018-08-04 by dlech