ev3dev/ev3dev #1142: FLL

dlech, 2018-08-04:
### What are you trying to do?

Let people know that ev3dev can be used for FIRST LEGO League in select regions this year, including my home state of Oklahoma.

### What did you expect to happen?

Some people will be very excited to hear this.

### What actually happened?

It is yet to be seen...

dlech, 2018-08-04:
So, FLL season is already in full swing (at least here in the USA). The challenge was released on 1 August. It will be interesting to see how many teams actually use ev3dev. Also it opens up the question: what is needed to make ev3dev "FLL ready"?

moriarty, 2018-08-04:
That sounds cool!
Can you provide a link to the rules?

http://www.firstlegoleague.org/challenge

# Throughout the season, teams need access to the following materials:
    # LEGO® MINDSTORMS Robot Set
    # Computer or tablet with software to program the robot. (Laptop is recommended.)

> with software to program the robot
So that doesn’t specifically say Lego Mindstorms Software like in former rules so ev3dev is allowed?

dlech, 2018-08-04:
This year, using other programming languages is being done on a per-region opt-in basis, so the general rules are the same as in past years. FLL coordinators in regions that have opted in will receive special rules. For example Ohio FLL posted [this on facebook](https://www.facebook.com/ohfll/photos/a.1413669698846922.1073741832.1385524211661471/2125504434330108/?type=3&theater).

WasabiFan, 2018-08-04:
Woah! I didn't know they were opening up the software restrictions. That's quite exciting!

jabrena, 2018-08-04:
Very good news!!!

sanjayseshan, 2018-08-05:
Yes. @arvindseshan and I are very exited for the opportunities this brings to teams. We are working with @dlech to create a chart comparing different programming languages to help teams make intelligent choices. Let's see how many teams pick ev3dev!

dwalton76, 2018-08-06:
That is very cool :)

sanjayseshan, 2018-08-09:
To make it easier for teams to install/setup ev3dev, we wrote a new lesson on ev3lessons.com: <a href="http://ev3lessons.com/en/ProgrammingLessons/beyond-ev3g/EV3Dev-intro.pdf">Lesson PDF</a>.

Thank you to @dlech for the feedback he gave on the lesson.

closed 2019-01-28 by dlech