ev3dev/ev3dev #1143: Network connection problem (USB)

xslendix, 2018-08-06:
### What are you trying to do?
I am trying to connect my EV3 to an internet connection using USB cable.

### What did you expect to happen?
To be able to use commands such as git, wget, apt etc.

### What actually happened?
I can connect to my EV3 using the local ip of its: 192.168.137.3 (Might change idk)

### What hardware and software are you using (including version numbers)?
WIN10 Laptop

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```
[image]
[image]

dlech, 2018-08-06:
It looks like you are trying to share your EV3 to your Wi-Fi rather than the other way around.

xslendix, 2018-08-06:
Oh, so, I was supposed to put WiFi on the ev3 and not ev3 on the wifi?

dlech, 2018-08-06:
right

xslendix, 2018-08-06:
so, on WiFi, I set it back to Ethernet and on EV3, I set it to WiFi?

xslendix, 2018-08-06:
let me see if it works

xslendix, 2018-08-06:
Yeah, it does not work and no longer cann connect to SSH

dlech, 2018-08-06:
Guess I was wrong. You should do whatever the tutorial says. What wasn't working exactly?

xslendix, 2018-08-07:
It simply does not have connection to internet

dlech, 2018-08-07:
Does the wired network connection on the EV3 say "connected" or "online"?

xslendix, 2018-08-07:
Yes

dlech, 2018-08-07:
That was an either-or question. It can't say both at the same time.

xslendix, 2018-08-07:
Oh. Connected

xslendix, 2018-08-07:
Sorry

dlech, 2018-08-07:
And you followed the tutorial exactly and didn't do any extra steps like selecting "Windows defaults" on the EV3?

xslendix, 2018-08-07:
Yes. Also, I tried other selections on the EV3

dlech, 2018-08-07:
Well, it looks like you were doing it right the first time. Usually if it stops working, unconfiguring Internet Connection Sharing and setting it back up is enough to get it working again.

staadecker, 2019-04-20:
@dlech I have the same issue. Any suggestions?

dlech, 2019-04-20:
What are the exact steps you performed? I just tested with a fresh install of the ev3dev-stretch-2019-03-03 image and Windows 10 1809 following [this tutorial](https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/) and it works. 

Do you have any security software that might be interfering?

staadecker, 2019-04-20:
Surprisingly it worked. I didn't change anything however this time it gave me the option to select which network connection to share the internet to. Thank you.

dlech, 2019-04-20:
Usually, disabling Internet Connection Sharing on Windows and enabling it again will fix connection problems.

n33r4j, 2020-01-28:
@xslendix Were you able to solve the issue?
I believe I'm stuck at the same point where it shows "connected" instead of "online" as in the tutorial, although I'm running a newer version (ev3dev-stretch-ev3-generic-2019-10-23).

dlech, 2020-01-28:
Are you using Bluetooth and Windows?

n33r4j, 2020-01-28:
Windows 10, and the EV3 is connected via it's USB cable to my laptop.

dlech, 2020-01-28:
That should work.

n33r4j, 2020-01-28:
Is it necessary that the 'state' on the EV3 should change to 'online' or is there a chance that it works even when it shows 'connected'?
What prompted me to check this is the inability to run commands such as 'sudo apt-get update' and cloning the 'ev3dev-c' repo.