ev3dev/ev3dev #1145: Arduino connection

xslendix, 2018-08-08:
### What are you trying to do?
I am trying to connect my Arduino to EV3 to communicate to each uther using PySerial.

### What did you expect to happen?
To communicate to each other

### What actually happened?
I actually gave an "Invalid COM port" so I don't know waht the USB port on the EV3 is...

### What hardware and software are you using (including version numbers)?
| Image file            | ev3dev-stretch-ev3-generic-2018-08-06 |
|-----------------------|---------------------------------------|
| Image file            | ev3dev-stretch-ev3-generic-2018-08-06 |
| Kernel version        | 4.14.61-ev3dev-2.2.2-ev3              |
| Brickman              | 0.10.0                                |
| BogoMIPS              | 148.88                                |
| Board                 | board0                                |
| BOARD_INFO_HW_REV     | 7                                     |
| BOARD_INFO_MODEL      | LEGO MINDSTORMS EV3                   |
| BOARD_INFO_ROM_REV    | 6                                     |
| BOARD_INFO_SERIAL_NUM | 00165345D5C8                          |
| BOARD_INFO_TYPE       | main                                  |

dlech, 2018-08-08:
ev3dev is Linux, so any info on using Arduino with Linux should help. For example, if it is a serial connection, I would expect it to be something like `/dev/ttyUSB0`.

xslendix, 2018-08-08:
Tried it, not working

xslendix, 2018-08-08:
I found stuff in /dev/serial/by-id a USB port and in by-name I found my Leonardo

JorgePe, 2018-08-10:
I guess you already figure it out but when you plug your Arduino you can find the serial connection that is associated to it with 'dmesg'.

You can also run
```dmsg -w```
then plug your Arduino and see what it shows.
it it creates a serial connection, it should say something like '/dev/ttyUSBx' or '/dev/ttyACMn' or '/dev/ttyS0' depending on the chipset used by your Arduino board.

jabrena, 2018-08-10:
You should not have any problem to have bidirectional communication with EV3 using EV3Dev as OS.

You have establish the serial comms with Python or Java without any problem.

Here you have an example to connect with an IMU mounted on Arduino:
https://github.com/ev3dev-lang-java/usb-devices/blob/master/src/main/java/ev3dev/sensors/arduino/bn055/BNO055.java

Juan Antonio