ev3dev/ev3dev #1147: Brickpi3 + raspberry 3B same value after sensor disconnected

adesito, 2018-08-09:
### What are you trying to do?

Hi. In my project involving the lego robo arm H25 (standar lego set). We have created a tornado 5.1 non blocking web server for API REST, to control the arm, but we need to check if a thermal sensor is connected or not. 

I know how to create and start up the sensors with brickpi3 (manually no auto), but if I disconnect the sensor I continue reciving last value forever... is there any kind of workaround to know if the sensor was disconnected? maybe if no value is receiving in that port, then you could get 0 or -1 or something like that?...date and time of files that we could check?

### What did you expect to happen?
If I disconnect a sensor no value or 0 (or -1 maybe) would be received.

### What actually happened?
I get forever, last value before the sensor was disconnected, for example if the last sensor value was 255 (1 decimal 25.5) and I disconnect the sensor, I get 255 forever (or until I reconnect the sensor and get values again).

### What hardware and software are you using (including version numbers)?

Image file:         ev3dev-stretch-rpi2-generic-2018-07-15
Kernel version:     4.14.56-ev3dev-2.2.1-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000a2c7df57
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=BF9020BD5150324336202020FF140E30
BOARD_INFO_TYPE=aux

ev3-python 1.2.0 and brikpi3

Regards!

dlech, 2018-08-09:
This is a firmware limitation of the BrickPi3, so you would need to talk to Dexter Industries about this.

The only way I can think to work around this is to change the port to `analog-nxt` mode and read the analog value, then change the mode back to the proper sensor. But, this will only work if the sensor you are using changes the voltage reading on pin 1.

What sensor are you using exactly?

adesito, 2018-08-09:
Hi... Thanksgiving forma the advice. Os the lego nxt thermal sensor... I
Talk yo Dexter and they toldos me that they don't support nxt thermal
sensor... So Usted test with your workaround.

Thank You so mucho
Regards.