ev3dev/ev3dev #1150: Sensor lag

Firebrand, 2018-08-12:
Using **ev3dev-stretch-ev3-generic-2018-08-06** on **Lego Mindstorms EV3**

I'm a newbie to ev3dev but it seems that there is quite a bit of delay in the sensor response times. Below is a simple test I ran with the ultrasonic sensor. In ev3dev python, when I run this and put my hand in front of the sensor it takes about 2 to 3 secs(!) to actually register any changes. In the mindstorms visual env it's milliseconds. Am I doing something wrong?

#!/usr/bin/env python3
from ev3dev2.display import *
from ev3dev2.fonts import *
from ev3dev2.sensor.lego import UltrasonicSensor

ultraSense = UltrasonicSensor()
ultraSense.mode = 'US-DIST-CM'
display = Display()

while ultraSense.value() > 100:
    display.clear()
    display.draw.text((10,10), str(ultraSense.value()), font=fonts.load('luBS14'))
    display.update()

dlech, 2018-08-12:
I think the problem is that you are calling `fonts.load()` inside the loop and it is re-loading the font from disk each time.

Firebrand, 2018-08-12:
Man, thanks so much! I'm super embarrassed by my question now... getting that font load statement out of the loop should've been the first thing to try.

Firebrand, 2018-08-12:
@dlech  Thanks for the suggestion dlech. Unfortunately, I completely removed the font loading statement and it still takes 3-4 secs to show the new value of the sensor (this happens with all sensors). This is my new code:

#!/usr/bin/env python3
from ev3dev2.display import *
from ev3dev2.fonts import *
from ev3dev2.sensor.lego import UltrasonicSensor

ultraSense = UltrasonicSensor()
ultraSense.mode = 'US-DIST-CM'
display = Display()

while True:
    display.clear()
    display.draw.text((10,10), str(ultraSense.value()))
    display.update()

dlech, 2018-08-12:
Do the values update faster if you view them using the Device Browser in Brickman?

Firebrand, 2018-08-12:
@dlech Yes. If I go to Device Browser > Sensors > Watch Values on the sensor I choose, the values update in less than a second as opposed to 3-4 seconds.

dlech, 2018-08-12:
Then there must be a problem in the python library.

cc: @ev3dev/python

WasabiFan, 2018-08-13:
In initial testing I can confirm the issue and have isolated it to the screen drawing. Looking into this now.

WasabiFan, 2018-08-13:
It looks like we dramatically regressed display update performance here: https://github.com/ev3dev/ev3dev-lang-python/commit/70b7b449824c996196ed5b092213b3306e7a69f2

The logic to update the display with 32bpp data [here](https://github.com/ev3dev/ev3dev-lang-python/blob/e2dd9a1180eed72e20a22f5f53744b06754b9048/ev3dev2/display.py#L310-L311) takes multiple seconds on the EV3. My inclination is to say that this is because our 32bpp image-to-byte conversion logic is implemented with Python list comprehensions whereas the Pillow equivalent (which we use for 1bpp) is [written natively in C](https://github.com/python-pillow/Pillow/blob/af552801c25002baaf3e0239206bd7bb395eaf3e/src/encode.c#L467).

dlech, 2018-08-13:
It looks like there is [XRGB in Pack.c](https://github.com/python-pillow/Pillow/blob/81c88b1ead5343d9778a82977c9c6c1150fd9738/src/libImaging/Pack.c#L560) if you can figure out how to use that.

dlech, 2018-08-13:
I have a feeling this might work:

    self.mmap[:] = self._img.tobytes(encoder_name='XRGB')

WasabiFan, 2018-08-14:
Built-in encoders appear to be defined here: https://github.com/python-pillow/Pillow/blob/90a94e7dae19f75363c769713b1b6d10eb16f613/src/_imaging.c#L3694. There's no XRGB encoder (or, an encoder for any other plain pixel data format it seems) so I expect I need to look into finding an API that lets us invoke one of the "packers" that you linked to. I'll take a look today or tomorrow probably.

WasabiFan, 2018-08-14:
I think the bottom line is that there is no existing packer defined which converts from any greyscale (single-channel) format to any color format. So while it is possible to invoke them, they will never be able to convert as we want. That being said, I think I can probably do a `merge` of the single-channel image with itself to produce a three-channel greyscale image and then operate on that.

Firebrand, 2018-08-14:
Cool. I wish I had more Python/C exp to help you guys out with the beautiful work you do. Another ev3dev2 release comin up soon?

WasabiFan, 2018-08-14:
I've opened ev3dev/ev3dev-lang-python#497 with my proposed fix. It gets screen drawing performance to an acceptable level, but that's still not nearly as fast as it was when we were able to directly ship the bits from the internal image buffer to the display framebuffer (as we did on ev3dev-jessie). @dlech do you have any other suggestions in this space, or is my change the best you think we are going to get?

WasabiFan, 2018-08-14:
> Another ev3dev2 release comin up soon?

That depends on a lot of factors, but yes I do intend to make another release soon. I'll probably hold off until I can include a couple other improvements that we have queued.