ev3dev/ev3dev #1151: Error reading ev3 motor position

JenniceDK, 2018-08-13:
Hi all,

I have attached a Python file for my Ev3, where I control motor movement.
This part funs fine.

At the end I have tried to read the motor position of motor "m" (defined in first part), and print he tacho value to the EV3 display, but it fails. The program crashes and the ev3 returns to main screen.

It's probably something "stupid" I have done... I just don't know what's wrong.

dlech, 2018-08-13:
Please answer all of the questions in the issue template:

```
### What are you trying to do?

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

<!-- Tip: copy and paste the output of `ev3dev-sysinfo -m` here -->

```

JenniceDK, 2018-08-13:
### What are you trying to do?

I am trying to read the motor rotary (tacho?) position, using a standard Large EV3 motor.
The program works until the while loop.

The program is between the >>>>><<<<<:

>>>>>
#!/usr/bin/env python3
# so that script can be run from Brickman

from ev3dev.ev3 import *
from time import sleep

m = LargeMotor('outB')

m.run_to_rel_pos(position_sp=360, speed_sp=90, stop_action="coast")

sleep(5)   # Give the motor time to move
m.run_to_rel_pos(position_sp=-360, speed_sp=90, stop_action="coast")

sleep(5)   # Give the motor time to move

while true:
    print(m.position())
    sleep(1)
<<<<<

### What did you expect to happen?
I hoped to get the position as a number, and print it on the EV3 screen.

### What actually happened?
The program crashed when getting to the command to read position.

### What hardware and software are you using (including version numbers)?
Python 3
Brickman 0.8.1
Ev3 brick revision 0006.

(Not sure what this is). Not sure if it's relevant. I think I may be missing an import/library or using the command wrong.

dlech, 2018-08-13:
Please try ev3dev-stretch. There are many improvements over ev3dev-jessie.

Also, your crash is probably caused by calling position as a method instead of as a property.

Try this:

    print(m.position)

JenniceDK, 2018-08-13:
Thank you, dlech :)

It was indeed the m.position part I had wrong. Thank you for helping and solving my problem.
It now works! 👍

closed 2018-08-13 by JenniceDK