ev3dev/ev3dev #1152: Driver for Mindsensors IR Temperature Sensor

donei34, 2018-08-18:
### What are you trying to do?
Use the mindsensors IR Temperature Sensor for EV3 or NXT with ev3dev
[http://www.mindsensors.com/products/170-ir-temperature-sensor-for-ev3-or-nxt](http://www.mindsensors.com/products/170-ir-temperature-sensor-for-ev3-or-nxt)

### What did you expect to happen?
A driver may exist somewhere?

### What actually happened?
A diver didn't exist

### What hardware and software are you using (including version numbers)?
NA

Would a driver for this sensor be able to be added / made (it uses I2C). In case it helps, here is the RobotC implementation of a driver [https://github.com/botbench/robotcdriversuite/blob/master/include/mindsensors-irthermometer.h](https://github.com/botbench/robotcdriversuite/blob/master/include/mindsensors-irthermometer.h)

dlech, 2018-08-18:
This sensor looks quite simple. Would you like to add the driver for it? Basically, you just need to copy an existing sensor, [like this one](https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/sensors/nxt_i2c_sensor_defs.c#L1942) and change the values based on the information from the users guide.

donei34, 2018-08-18:
Thanks for the quick reply!!!!
Honestly, I wouldn't know how exactly to modify it. I could probably try the basics, but it doesn't have commands or modes AFAIK. Also, not entirely confident on forking and creating pull requests.

dlech, 2018-08-18:
It will have 4 modes. `AMBIENT-C`, `TARGET-C`, `AMBIENT-F` and `TARGET-F` (or 6 if you think we need modes that read both ambient and target at the same time.

You can actually modify the file right in the web browser.  Just click the edit icon (looks like a pencil) at the [top of this page](https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/sensors/nxt_i2c_sensor_defs.c).

donei34, 2018-08-18:
Oh, ok, thanks. I'll see where I get to in creating a driver. What data types are supported?

dlech, 2018-08-18:
the basic types are signed 8, 16 and 32-bit integers (little-endian). `LEGO_SENSOR_DATA_S16` is probably what we want here.

donei34, 2018-08-22:
Hi, 
I haven't really done too much about it, but noticed that the sensor has two registers for each value, and I don't known how to implement this in the driver defs.

dlech, 2018-08-22:
The two values are just the two bytes of the 16 byte value. You don't have to do anything special other than specify the data type I already mentioned.

donei34, 2018-08-25:
What do I do for this section:
```
.i2c_mode_info	= (const struct nxt_i2c_sensor_mode_info[]) {
			[0] = {
				.read_data_reg	= 0x42,
			},
			[1] = {
				.read_data_reg	= 0x46,
			},
			[2] = {
				.read_data_reg	= 0x4A,
			},
			[3] = {
				.read_data_reg	= 0x42,
			},
		},
```
Since there are two registers for each mode

donei34, 2018-08-25:
Also there are no commands, so can I just delete this section:
`.cmd_info	= (const struct lego_sensor_cmd_info[]) {
			[0] = {
				/**
				 * @description: Reset angle values
				 */
				.name = "RESET",
			}
		},
		.i2c_cmd_info	= (const struct nxt_i2c_sensor_cmd_info[]) {
			[0] = {
				.cmd_reg	= 0x41,
				.cmd_data	= 'r',
			},
		}`

dlech, 2018-08-25:
> Since there are two registers for each mode

The address will be the lower of the two and the data type will need to be `LEGO_SENSOR_DATA_S16`, which is two bytes. Also, if I recall correctly, `decimals` needs to be set to `2`. There are some comments towards the top of the file that explain the possible fields.

> so can I just delete this section

correct

donei34, 2018-08-26:
Ok, that file modification is done. What else do I need to modify to get the sensor working?

dlech, 2018-08-26:
Can you go ahead and open a pull request so I can have a look at the code? Also, if you could [build your own kernel](https://github.com/ev3dev/ev3dev-buildscripts/#ev3dev-buildscripts) and test it, that would be awesome. Otherwise, we'll have to wait for the next kernel release, whenever that is.

donei34, 2018-09-05:
Hi,
Sorry for the long wait. I've created a pull request for you to have a look at
Thanks

dlech, 2018-09-05:
I don't see the pull request. Where did you create it?

donei34, 2018-09-05:
Whelp, the joys of being new to github. Sigh. I'll recreate it shortly

donei34, 2018-09-05:
Created (again)

dlech, 2018-09-08:
It looks like there is one thing we missed in https://github.com/ev3dev/lego-linux-drivers/pull/70. The `product_id` field must exactly match what is programmed in the sensor and it can only be 8 characters max, so I know the value we have is not correct. 

Can you please run the following command on your EV3 with the sensor attached to port 4:

    sudo i2cdump -f -y -r 0x00-0x17 6 0x15

donei34, 2018-09-09:
Here is the result:
```
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 56 31 2e 30 32 00 00 00 6d 6e 64 73 6e 73 72 73    V1.02...mndsnsrs
10: 49 52 54 68 65 72 6d 6f                            IRThermo
```
What do I do about the pull request / modification then, just change it in my branch and create a new request?

dlech, 2018-09-09:
It is just one line that needs to be changed, so I can take care of it. For future reference though, it a situation like this, you would need to create a new pull request since the other one has been merged already.

donei34, 2018-09-09:
Ahh, ok. Thank you so much for your help!!!!!!