ev3dev/ev3dev #1153: Doubt about the notes to increase the CPU speed for EV3 on Debian Stretch

jabrena, 2018-08-20:
Hi @dlech,

I am testing the new parameter `cpufreq`

```
# uncomment to increase CPU frequency
# 375MHz is rated speed at available voltage (1.2V)
# 456MHz is max rated speed of CPU (supposed to have 1.3V for this)
cpufreq=375
```

How to know the Brick voltage?

Juan Antonio

dlech, 2018-08-20:
By looking at the schematic. CVDD has a fixed 1.2V regulator.

jabrena, 2018-08-20:
So in what case, it is possible to set `cpufreq` with the value `456`? When the brick is connected to a charger or when the brick use the rechargable battery?

I am testing with value `375` and I am observing a nice performance improvement. For example, when you try to install a new Debian package, the process to create the tree of dependencies is a classic performance issue on EV3 but now, the process goes faster.

dlech, 2018-08-20:
read the blog post