ev3dev/ev3dev #1154: Internet via Bluetooth only Connected - not Online

KWrne, 2018-08-20:
## Internet via Bluetooth only Connected - not Online

### What are you trying to do?

I'm trying to connect my EV3 to the internet via Bluetooth

### What did you expect to happen?

I expected the status to say online and allow me to access PuTTY

### What actually happened?

Instead, it just said connected and I can't connect to PuTTY. My DHCP doesnt work properly (I don't think so at least), I run Windows Default because of this and it is the only way it connects partially to my laptop. I have looked at other issues thoroughly but they haven't been able to solve my issue.

### What hardware and software are you using (including version numbers)?

EV3 Brick
ev3dev-stretch-ev3-generic-2018-08-06

dlech, 2018-08-20:
Don't use "Windows Defaults" - it will only hide the problem, use "DHCP".

Usually turning of Windows Internet Connection sharing and turning it back on is enough to fix the problem.

FalconFX9, 2019-03-31:
Hello, I have the same issue, and from my research, it seems that on Windows 10, sharing an internet connection to a Bluetooth PAN (that my ev3 is connected to), is simply impossible. Unless anyone has a solution ? (Note: the ev3 connects to the internet via usb, just bluetooth doesn't work).

staadecker, 2019-04-20:
Yup same issue as @FalconFX9 . This means I cannot run `sudo apt` commands.

dlech, 2019-04-20:
It seems that Internet Connection Sharing for Bluetooth is broken on at least Windows 10 1809. If you need an Internet connection to your EV3, we recommend using USB or Wi-Fi. Even if Bluetooth was working, it tends to crash if you try to download more than a few MB (i.e. `apt install` only works for small packages via Bluetooth).

staadecker, 2019-04-20:
Unfortunately I have the same issue over USB. I can SSH but when I run `ping google.ca` it doesn't work.

dlech, 2019-04-20:
Please open a new issue or find an open existing issue for USB.

dlech, 2019-04-20:
https://answers.microsoft.com/en-us/windows/forum/windows_7-networking/internet-connection-sharing-bluetooth-network/b09b28a6-0673-4714-bbe0-845f014eb94f

FalconFX9, 2019-04-20:
@dlech So does this confirm there is no way to do this on the current Windows 10 build without doing some rather complicated/shady stuff (found some users getting WIFI over PAN by using Toshiba bluetooth Stack, on windows 10, but that requires disabling driver signature and a bunch of other stuff) ?

dlech, 2019-04-20:
I'm certainly not having any luck getting it working. Documentation on how Windows Internet Connection Sharing works seems to be nearly non-existent. This is the best I have found so far: https://docs.microsoft.com/en-us/windows/desktop/nativewifi/using-hosted-network-and-internet-connection-sharing

And documentation on the Window 10 Mobile Hotspot feature seems to be completely non-existent. Although I assume it just uses the existing ICS under the hood.

FalconFX9, 2019-04-20:
Yeah, from what I've read, Microsoft killed all native support for WIFI over BT PAN from Vista onwards, and the only option seems to be BT Stack. It's too bad the current bluetooth internet sharing option on Windows 10 just cuts BT connection to the ev3 instead of actually sharing internet to the ev3. I might try installing BT Stack when I get the time to try it out.