ev3dev/ev3dev #1155: BrickPI 3 + Rapsberry PI

Negociation, 2018-08-21:
### What are you trying to do?
Hi, i want to merge from EV3 Brick to Rapsberry Pi  , but i have some doubts , first i checked on website i olny can find a ISO for Rapsberry PI 2 , thats mean i cant work with Rpasberry PI 3 ?

- I Checked the Dexter Industries Website they only haves BrickPI 3, everone knows if BrickPI 3 as compatible with EV3DEV ?

Please i need to know that for my Final Paper on college , i probably have to buy on next week...

jabrena, 2018-08-21:
It is compatible and running

Negociation, 2018-08-21:
So can i run with Rapsberry PI 3 + BrickPI 3 without issues , right ?

jabrena, 2018-08-21:
Yes, many months ago from OS layer, the support was implemented so you could develop. Other stuff is the language that want to use. If you use/like Python maybe I should add @dwalton76 in the loop. 
In relation to Java, recently we added it: 
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/518

@mingan666 is helping us with final testing step for the combination: BrickPi3 + Raspberry Pi 3 Model B+ + Java.

**Note:** For other languages, I don´t know the real status.

Good night.

Cheers

Juan Antonio

Negociation, 2018-08-21:
Im building a self deiving car with Convolution Neural Newtworks , so im using Python because Tensorflow.... Now im building my dataset so i can still using LEGO but when i starts the test of application I probably need more processing power i guess , that's what's I want to using the raspberry instead...

jabrena, 2018-08-21:
Yes, definitely here, you will find good guys to collaborate.

I recommend to review this project in parallel to get some ideas.
http://www.donkeycar.com/

To manage the sensor and actuators from BrickPi3 here.

Cheers

Juan Antonio

Negociation, 2018-08-21:
Tanks bro, i'll look later... 

For Now my robot uses:
- Lego Gyro Sensor for Fix the Foward Steering (Very Dummie but...)
- 2 Large Motors
- Color Sensor in Col-Ambient
- Ultrasonic Sensor for get the front proximity
- Infrared Sensor in IR-REMOTE to Control
- 2 Remote Controls
- Android Smartphone on Front with TCP Implementation to Recover Video

Look:
[image]

I Want to Put front LEDs with GPIO after i replace for Rapsberry...

Negociation, 2018-08-24:
Its a requirement port my code without change platform dude... And ev3dev with Python as a excelent choice... I checked and if i use the OS from dexter i have to change a lot of things on my code...