ev3dev/ev3dev #1156: linux-image-4.14.61-ev3dev-2.2.2-ev3 not reporting version correctly?

kortschak, 2018-08-23:
### What are you trying to do?
Upgrade to latest (linux-image-4.14.61-ev3dev-2.2.2-ev3) kernel.

### What did you expect to happen?
A successful upgrade

### What actually happened?
An upgrade that reports its version as 4.14.35-ev3dev-2.0.0-ev3 despite not having that kernel installed.

```
~ $ ssh robot@ev3dev.local
Password: 
Linux ev3dev 4.14.35-ev3dev-2.0.0-ev3 #1 PREEMPT Sun Apr 22 13:59:21 CDT 2018 armv5tejl
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
 |  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
Last login: Thu Aug 23 11:07:41 2018 from 192.168.178.24
robot@ev3dev:~$ uname -a
Linux ev3dev 4.14.35-ev3dev-2.0.0-ev3 #1 PREEMPT Sun Apr 22 13:59:21 CDT 2018 armv5tejl GNU/Linux
robot@ev3dev:~$ dpkg -l 'linux-image-*'
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                            Version                      Architecture                 Description
+++-===============================================-============================-============================-===================================================================================================
un  linux-image-4.14                                <none>                       <none>                       (no description available)
ii  linux-image-4.14.61-ev3dev-2.2.2-ev3            1                            armel                        Linux kernel binary image for version 4.14.61-ev3dev-2.2.2-ev3
un  linux-image-4.14.61-ev3dev-2.2.2-ev3-dbg        <none>                       <none>                       (no description available)
ii  linux-image-ev3dev-ev3                          51                           armel                        Linux kernel for ev3dev (metapackage)
```

### What hardware and software are you using (including version numbers)?

Well, that's the funny thing.

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2017-09-30
Kernel version:     4.14.35-ev3dev-2.0.0-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534CB28D
BOARD_INFO_TYPE=main
```

dlech, 2018-08-23:
This is probably a duplicate of #612. However, you are probably missing out on some things by not updating the full image to the 2018-08-06 image.

kortschak, 2018-08-23:
I don't think it's exactly the same as #612 for a couple of reasons, it's not a dist-upgrade and I'm prettty sure I saw flash-kernel run.

It's seems very odd to me that a new image is needed when I'm merely updates something within a debian major version.

dlech, 2018-08-23:
Did you use `apt upgrade` or `apt-get upgrade`? `apt update` is the equivalent of `apt-get upgrade && apt-get dist-upgrade`. And the bug is that flash-kernel runs (sort of) but doesn't actually put the new kernel where it needs to be.

The reason for needing the new image is there were changes made to the image that can't be packaged, like a new bootloader.

kortschak, 2018-08-23:
OK, when I manually run flash-kernel I see this:
```
robot@ev3dev:~$ sudo flash-kernel linux-image-4.14.61-ev3dev-2.2.2-ev3
Using DTB: da850-lego-ev3.dtb
Alternate DTBs: da850-lego-ev3-lms2012.dtb da850-lego-ev3-adafruit18.dtb da850-lego-ev3-color.dtb
Couldn't find DTB da850-lego-ev3-lms2012.dtb in /usr/lib/linux-image-linux-image-4.14.61-ev3dev-2.2.2-ev3 or /etc/flash-kernel/dtbs
```

This did not appear during the apt upgrade. The file exists according to `dpkg -S`:
```
robot@ev3dev:~$ dpkg -S da850-lego-ev3-lms2012.dtb
linux-image-4.14.61-ev3dev-2.2.2-ev3: /usr/lib/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-lms2012.dtb
```

but is not in the correct place:
```
robot@ev3dev:~$ ls /usr/lib/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-lms2012.dtb
/usr/lib/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-lms2012.dtb
robot@ev3dev:~$ ls /usr/lib/linux-image-linux-image-4.14.61-ev3dev-2.2.2-ev3
ls: cannot access '/usr/lib/linux-image-linux-image-4.14.61-ev3dev-2.2.2-ev3': No such file or directory
```

Presumably either the location it is looking is incorrect or the file should have been symlinked/copied to the place where it's looking now, flash-kernel is not planning on using it and so should be quiet. What is more weird is that if I do symlink the /usr/lib/linux-image-4.14.61-ev3dev-2.2.2-ev3 to /usr/lib/linux-image-linux-image-4.14.61-ev3dev-2.2.2-ev3 it can still not find the dtb file, though it is clearly readable from there. A potentially interesting datapoint here is that the "linux-image" prefix is repeated in the flash-kernel-queried directory name, which seems like it shouldn't be.

kortschak, 2018-08-23:
I used `apt upgrade`.

kortschak, 2018-08-23:
I didn't realise that apt upgrade does a dist-upgrade. That's horrible.

kortschak, 2018-08-23:
I was able to get flash-kernel to apparently succeed by actually copying the directory rather than symlinking
```
robot@ev3dev:~$ sudo flash-kernel linux-image-4.14.61-ev3dev-2.2.2-ev3
Using DTB: da850-lego-ev3.dtb
Alternate DTBs: da850-lego-ev3-lms2012.dtb da850-lego-ev3-adafruit18.dtb da850-lego-ev3-color.dtb
Installing /usr/lib/linux-image-linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-lms2012.dtb into /boot/dtbs/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-lms2012.dtb
Installing new da850-lego-ev3-lms2012.dtb.
Installing /usr/lib/linux-image-linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-adafruit18.dtb into /boot/dtbs/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-adafruit18.dtb
Installing new da850-lego-ev3-adafruit18.dtb.
Installing /usr/lib/linux-image-linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-color.dtb into /boot/dtbs/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-color.dtb
Installing new da850-lego-ev3-color.dtb.
Installing /usr/lib/linux-image-linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3.dtb into /boot/dtbs/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3.dtb
Installing new da850-lego-ev3.dtb.
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "en_AU.UTF-8"
    are supported and installed on your system.
	LANG = "en_AU.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
perl: warning: Falling back to the standard locale ("C").
Ignoring old or unknown version linux-image-4.14.61-ev3dev-2.2.2-ev3 (latest is 4.14.61-ev3dev-2.2.2-ev3)
Use --force if you want version linux-image-4.14.61-ev3dev-2.2.2-ev3.
Installing /usr/lib/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-lms2012.dtb into /boot/dtbs/4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-lms2012.dtb
Taking backup of da850-lego-ev3-lms2012.dtb.
Installing new da850-lego-ev3-lms2012.dtb.
Installing /usr/lib/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-adafruit18.dtb into /boot/dtbs/4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-adafruit18.dtb
Taking backup of da850-lego-ev3-adafruit18.dtb.
Installing new da850-lego-ev3-adafruit18.dtb.
Installing /usr/lib/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-color.dtb into /boot/dtbs/4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3-color.dtb
Taking backup of da850-lego-ev3-color.dtb.
Installing new da850-lego-ev3-color.dtb.
Installing /usr/lib/linux-image-4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3.dtb into /boot/dtbs/4.14.61-ev3dev-2.2.2-ev3/da850-lego-ev3.dtb
Taking backup of da850-lego-ev3.dtb.
Installing new da850-lego-ev3.dtb.
flash-kernel: installing version 4.14.61-ev3dev-2.2.2-ev3
Generating boot script u-boot image... done.
Taking backup of boot.scr.
Installing new boot.scr.
```
Though not without errors and not in a way that has fixed the problem. On next reboot, uname returns the same 2.0.0 kernel version.

dlech, 2018-08-23:
Run `flash-kernel` without any arguments. `linux-image-` is not part of the kernel name, so it breaks things when you specify that as the kernel name.

dlech, 2018-08-23:
However, the new bootloader is probably required to make this work too.

If you really must insist on not flashing a new image, grab these files and throw them in the FAT partition (`/boot/flash`). https://github.com/ev3dev/u-boot/releases/tag/v2018.07-rc0-ev3dev2

There will probably still be other things that are broken though.

kortschak, 2018-08-23:
I'm going to give up on this and get a new image.

closed 2018-08-23 by kortschak