ev3dev/ev3dev #1157: Cross compiling Micropython using docker: Problem with axtls

laurensvalk, 2018-08-23:
I am able to cross-compile MicroPython and run for ev3dev using docker using

```
make CROSS_COMPILE=arm-linux-gnueabi-
``` 
But it works only if I disable the ``axtls`` module  [here](https://github.com/ev3dev/micropython/blob/d96c112c787e73f352634ba56979da3fabc07a37/ports/unix/mpconfigport.mk#L26).

The readme states that I have to ```make axtls``` to be able to include it. I tried
```
make axtls CROSS_COMPILE=arm-linux-gnueabi-
```

But it fails with 

```
make[2]: gcc: Command not found
```
So, I'm clearly not setting the compiler correctly here. Any pointers on how to do this correctly?

I added ``$(info $$CC is [${CC}])`` to the [make file](https://github.com/ev3dev/micropython/blob/ev3dev-stretch/ports/unix/Makefile) as follows.

```
$(info $$CC is [${CC}])
axtls: $(BUILD)/libaxtls.a

$(BUILD)/libaxtls.a: $(TOP)/lib/axtls/README | $(OBJ_DIRS)
	cd $(TOP)/lib/axtls; cp config/upyconfig config/.config
	cd $(TOP)/lib/axtls; $(MAKE) oldconfig -B
	cd $(TOP)/lib/axtls; $(MAKE) clean
	cd $(TOP)/lib/axtls; $(MAKE) all CC="$(CC)" LD="$(LD)"
	cp $(TOP)/lib/axtls/_stage/libaxtls.a $@
```
This does correctly state ```arm-linux-gnueabi-gcc``` in the resulting output below. But it does not appear to get passed on to the other make files.

```
compiler@536504150fda:/micropython/ports/unix$ make axtls CROSS_COMPILE=arm-linux-gnueabi-
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
$CC is [arm-linux-gnueabi-gcc]
cd ../../lib/axtls; cp config/upyconfig config/.config
cd ../../lib/axtls; make oldconfig -B
make[1]: Entering directory '/micropython/lib/axtls'
make -C config/scripts/config conf
make[2]: Entering directory '/micropython/lib/axtls/config/scripts/config'
/bin/sh: 1: gcc: not found
/bin/sh: 1: gcc: not found
/bin/sh: 1: gcc: not found
cp ../../scripts/config/zconf.tab.h_shipped zconf.tab.h
gcc -Wall -Wstrict-prototypes -O2 -fomit-frame-pointer -I. -c ../../scripts/config/conf.c -o conf.o
make[2]: gcc: Command not found
Makefile:59: recipe for target 'conf.o' failed
make[2]: *** [conf.o] Error 127
make[2]: Leaving directory '/micropython/lib/axtls/config/scripts/config'
Makefile:140: recipe for target 'config/scripts/config/conf' failed
make[1]: *** [config/scripts/config/conf] Error 2
make[1]: Leaving directory '/micropython/lib/axtls'
Makefile:288: recipe for target 'build/libaxtls.a' failed
make: *** [build/libaxtls.a] Error 2
```

dlech, 2018-08-23:
Which docker image are you using?

Perhaps another command besides `gcc` is not found. According to the [debian package](https://github.com/ev3dev/micropython/blob/ev3dev-stretch/debian/control#L5) there are some build dependencies that might not be installed. `libffi-dev, pkg-config, python, python3`

laurensvalk, 2018-08-23:
Image:
```
REPOSITORY                    TAG                 IMAGE ID            CREATED             SIZE
ev3dev/debian-stretch-cross   latest              8e3b0aa8f243        5 months ago        732MB
```
With ```libffi-dev, pkg-config, python, python3``` installed.

(MicroPython cross-compiles and runs fine if I exclude ```axtls```)

I can probably continue without it for now, but I was wondering if you encountered this.

I'll delve further into the other makefiles tomorrow. 

I'm currently on [this commit](https://github.com/micropython/micropython/commit/652a58698edadfe9b587324197e6f922790cf05f) of the MicroPython repository. I should probably try again with ev3dev/micropython.

laurensvalk, 2018-08-23:
> I assume you did remember to update git submodules? I haven't used docker
> to cross compile micropython before.

I think so, but I'll double check again tomorrow.

Ah, ok. Maybe I'll have to look into how the package is normally built
then.

laurensvalk, 2018-08-24:
Not the reason for this problem, but this [recent commit message](https://github.com/pfalcon/micropython/commit/ab78fe0eb9997622e40d412d5f7d5e9b1a91e47a) shows that this type of stuff may still be in progress. I'm adding it here just for reference.

laurensvalk, 2018-08-24:
I think it might be working now by adapting the CC settings in few places. I'll get back later.

laurensvalk, 2018-08-24:
You have to set ``HOSTCC=arm-linux-gnueabi-gcc`` in addition to ``CROSS_COMPILE=arm-linux-gnueabi-``
```
make axtls CROSS_COMPILE=arm-linux-gnueabi- HOSTCC=arm-linux-gnueabi-gcc
```
We could fix that in one of the makefiles upstream, but this is fine for now.

closed 2018-08-24 by laurensvalk

laurensvalk, 2018-08-24:
In fact, [this change](https://github.com/micropython/micropython/compare/master...laurensvalk:patch-1) would do. But I'm not sure that's a clean way to do it?