ev3dev/ev3dev #1159: Using the UART of the Input Ports of EV3

JorgePe, 2018-08-29:
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-05-24
Kernel version:     4.14.61-ev3dev-2.2.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534F4DD3
BOARD_INFO_TYPE=main
```

I've been dealing with the LEGO Interactive Motor (the motor from the BOOST set, with an encoder) and I can now read it with just a FTDI adapter, very similar setup to the article @dlech once wrote about using LEGO sensors in linux systems (TX, RX, GND, 3V3 and 115200 8N1)

Now I am trying to read the BOOST motor from EV3. Not because of the motor itself but because this might be useful later if all new Power Functions 2 / Powered Up sensor sensors use this same serial protocol.

The Motor has an internal microcontroller (a STM8S103F or similar).
The operating voltage is 2.95 to 5.5 V so I decided to connect the motor directly to an EV3 input port. I stripped a EV3 cable, connected GND, VCC, TX, RX, changed mode of IN1 to "other-uart" and tried my scripts with '/dev/ttyS1'... and only see sequences of '0'.

The motor still works, it's not an electrical problem.

I thought that perhaps I'm missing something on the UART configuration of the Input Port or the voltage levels so tried with the FTDI adapter instead ('/dev/ttyUSB0'). Also just zeros.

But this very same setup still works on Ubuntu laptop, I just plug/unplug the USB cable back and forth.

Is there something in ev3dev that might interfere with tty devices or might require different  settings?

I'm using "stty -F /dev/[ttydevice] 115200 raw" and it works in Ubuntu. Also compared the settings for both tty devices with stty and they look identical.

I'm using 'jpnevulator' to display the received data. It's an obscure program but allows me to see the data better than minicom, picocom or screen.

Any suggestions?

dlech, 2018-08-29:
I don't know what kind of buffering the FTDI has, but the built-in UARTs on the EV3 only have a 16 byte hardware buffer. At 115200 baud with no flow control, the EV3 has a really hard time keeping up with servicing interrupts before the hardware buffer overflows, especially in newer kernels (see #679 and some of the issues linked to that one).

If the sensors send data in bursts less that 16 bytes at a slow enough interval between bursts, this won't be a problem though. And it should be independent of transmitting data (I think).

What does your script look like?

JorgePe, 2018-08-29:
I believe the FTDI has a 62 bytes buffer and a 4k USB buffer... I'm probably filling it with my silly warm up:

```
#!/usr/bin/env bash

echo "Lots of 0"
for i in {1..50000}
do
   echo -e -n "\x00" > /dev/ttyUSB3
done

echo "Init"
echo -e -n "\x54\x22\x00\x10\x20\xB9" > /dev/ttyUSB3

for i in {1..50000}
do
    echo -e -n "\x02" > /dev/ttyUSB3
    sleep 0.1
done
```

This is on ubuntu. On ev3dev I use a version with the correct tty device and the number of initial zeros reduced to 1000.

After being initialized with that magic 6 byte string the motor keep sending 10 bytes every 0.1 seconds as long as I keep sending it a '02' every 0.1 seconds to keep it alive. If I don't do this it sends back some sort of warning and shortly gives up and needs to be reinitiated.

Recharging my EV3 but will try again soon, thanks for the tip.

JorgePe, 2018-08-31:
A better python script here:
https://github.com/JorgePe/re_interactivemotor/blob/master/initialcode/listen02.py

Still not using threads but understand a little better the motor and take two different actions if the motor isn't talking:

When the motor begs for initialization resend the Init string.
When motor is just sending zeros for too long force a resync (the voodoo part of sending lots of zeros).

It seems to work (not tried enough but until now never failed).  
For now it just works on my laptop, sorry.