ev3dev/ev3dev #1160: brightness level doesn't work (ev3dev-stretch, Lego EV3, LEDS)

xdsl, 2018-08-30:
### I try to change the brightness level:
1. echo 255 > /sys/class/leds/led0:green:brick-status/brightness 
2. echo 100 > /sys/class/leds/led0:green:brick-status/brightness 
3. echo 10 > /sys/class/leds/led0:green:brick-status/brightness 

### The result of all options is identical: the brightness level remains the maximum. 

### System:
Image file:         ev3dev-stretch-ev3-generic-2018-08-06
Kernel version:     4.14.61-ev3dev-2.2.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534C8680
BOARD_INFO_TYPE=main

jabrena, 2018-08-30:
Good morning,

What is the Debian Stretch image that you used?

xdsl, 2018-08-30:
https://github.com/ev3dev/ev3dev/releases/download/ev3dev-stretch-2018-08-06/ev3dev-stretch-ev3-generic-2018-08-06.zip

jabrena, 2018-08-30:
I reviewed your paths and they are ok:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/feature/%23443/src/main/resources/stretch.properties#L45

I will test it today.

Juan Antonio

xdsl, 2018-08-30:
paths is ok, but  brightness level doesn't change
if you send zero, the led will turn off
if you send nonzero, the led will light up on maximum brightness

dlech, 2018-08-30:
The brightness feature was removed in ev3dev-stretch.

WasabiFan, 2018-08-30:
Woah, really? Was that to avoid the overhead of bit-banging PWM?

xdsl, 2018-08-30:
Very bad. This is a regression.

dlech, 2018-08-30:
The pwm-gpio driver is not not in the mainline kernel, and I didn't port it to the newer kernel. However, now that we have use of one of the PRUs we could actually make a better PWM using that.