ev3dev/ev3dev #1161: Using hitechnic sensor multiplexer

lvjonok, 2018-08-30:
I have a hitechnic sensor multiplexer and want to read values from sensors that connected. So, can you give me example of reading information from hitechnic color sensor v2 that connected through multiplexer?
I tried to read info from Device Browser, but i couldnt do it.

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-08-06
Kernel version:     4.14.61-ev3dev-2.2.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535159F2
BOARD_INFO_TYPE=main
```

dlech, 2018-08-30:
What programming language are you using?

Have you looked at http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#ht-nxt-smux?

Since the HiTechnic Color Sensor V2 is on the list of automatically detectable sensors, you should be able to use the `DETECT` command to tell the multiplexer to detect sensors. You might need a short delay in you program while it does that. After that, the sensor should show up. You can use the Device Browser on the EV3 to find the address of the sensor.

dlech, 2018-08-30:
If the color sensor was already in the menu, then you shouldn't have to do anything special to use it other than you will need to look at the address and specify that in your program.

You can't use any EV3 sensors with this multiplexer. It only works with NXT analog and NXT I2C sensors. mindsensors.com makes an EV3 sensor multiplexer if you need one.

lvjonok, 2018-08-30:
When i plug in hitechnic color sensor, this mux port dissapears from Device Browser and when i send command DETECT from Device Browser nothing happens.

lvjonok, 2018-08-31:
So, when i try to use for example NXT light sensor, i write this code:
sens = LightSensor(address = "")
print(sens.value())
and address is what I see in Device Browser, and it doesnt works, it return an error that this device is not connected.

dlech, 2018-08-31:
I wasn't able to get automatic detection working either. However, you can manually configure the ports. See http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/muxs.html#ht-nxt-smux-port-mode-info.

In python, there is a class for ports (I don't remember the name off-hand). Then you set the mode `mux1.mode = 'analog'` and then tell it what kind of sensor is attached `mux1.set_device('lego-nxt-light`). Then you probably need a delay while the sensor is set up in the kernel `time.sleep(0.5)` and then you can create the `LightSensor` object.

lvjonok, 2018-08-31:
As I learned class for ports in Python named LegoPort, when I initialize mux1, I need to write address for multiplexer from Device Browser?

lvjonok, 2018-08-31:
`mux1 = LegoPort(address='i2c-legoev34:i2c8')
mux1.mode = 'analog'
mux1.set_device('lego-nxt-light')
sleep(1)
lS = LightSensor('i2c-legoev34:i2c8:mux1')`

This code returns an error:
Traceback (most recent call last):
  File "./new_moving.py", line 919, in <module>
    mux1.set_device('lego-nxt-light')
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 3073, in set_device
    raise Exception("set_device is a write-only property!")
Exception: set_device is a write-only property!

dlech, 2018-08-31:
It looks like `set_device` is a property for some reason. Try `mux1.set_device = 'lego-nxt-light'` instead.

lvjonok, 2018-08-31:
mux1 = LegoPort(address='i2c-legoev34:i2c8:mux1')
mux1.mode = 'analog'
mux1.set_device = 'lego-nxt-light'
sleep(1)
lS = LightSensor('i2c-legoev34:i2c8:mux1')
print(lS.value(0))

This is code for using NXT light sensor. Thank you for help!

closed 2018-08-31 by lvjonok