ev3dev/ev3dev #1163: Can't get motors to run on BrickPi3

jfcloutier, 2018-08-31:
### What are you trying to do?
Control EV3 motors from the console on a BrickPi3 setup

### What did you expect to happen?
The motors would run

### What actually happened?
Nothing

### What hardware and software are you using (including version numbers)?

Image file:         ev3dev-stretch-rpi2-generic-2018-08-06
Kernel version:     4.14.59-ev3dev-2.2.2-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000b98b6b55
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.3
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=F115113D515035524E4A2020FF07172E
BOARD_INFO_TYPE=aux

### 

What I did:
```
robot@ev3dev:/sys/class/lego-port/port4$ ls
address  driver_name  modes  set_device                status     uevent
device   mode         power  spi0.1:MA:lego-nxt-motor  subsystem

robot@ev3dev:/sys/class/tacho-motor/motor4$ ls
address        device         hold_pid   position_sp   speed      stop_action   uevent
command        driver_name    max_speed  power         speed_pid  stop_actions
commands       duty_cycle     polarity   ramp_down_sp  speed_sp   subsystem
count_per_rot  duty_cycle_sp  position   ramp_up_sp    state      time_sp

robot@ev3dev:/sys/class/tacho-motor/motor4$ echo reset > command
robot@ev3dev:/sys/class/tacho-motor/motor4$ echo 100 > duty_cycle_sp
robot@ev3dev:/sys/class/tacho-motor/motor4$ echo 500 > speed_sp
robot@ev3dev:/sys/class/tacho-motor/motor4$ echo run-forever > command
```
==> Nothing happens
```
robot@ev3dev:/sys/class/tacho-motor/motor4$ while true; do echo -en "\033[0G$(cat speed)   "; done
```
==> shows the speed changing when turning the motor by hand

jfcloutier, 2018-08-31:
BTW the sensors (touch, color, IR US) are all working fine.

jfcloutier, 2018-08-31:
I updated the brickpi3 firmware to no avail.

```
robot@ev3dev:/sys/class/lego-port/port4$ for f in /sys/class/tacho-motor/*; do echo -n "$f: "; cat $f/address; done
/sys/class/tacho-motor/motor0: spi0.1:MA
/sys/class/tacho-motor/motor1: spi0.1:MB
/sys/class/tacho-motor/motor2: spi0.1:MC
/sys/class/tacho-motor/motor3: spi0.1:MD
```
```
robot@ev3dev:/sys/class/tacho-motor$ ls
motor0  motor1  motor2  motor3
robot@ev3dev:/sys/class/tacho-motor$ cd motor0
robot@ev3dev:/sys/class/tacho-motor/motor0$ echo reset > command
robot@ev3dev:/sys/class/tacho-motor/motor0$ echo 500 > speed_sp
robot@ev3dev:/sys/class/tacho-motor/motor0$ echo run-forever > command
robot@ev3dev:/sys/class/tacho-motor/motor0$ echo 100 > duty_cycle_sp
robot@ev3dev:/sys/class/tacho-motor/motor0$ echo run-forever > command
```
Motor does not run.
```
robot@ev3dev:/sys/class/tacho-motor/motor0$ while true; do echo -en "\033[0G$(cat speed)   "; done
```
shows speed changing when manually turning the motor.

dlech, 2018-08-31:
does `echo run-direct > command` work? (`duty_cycle_sp` only affects this command and not any other command)

jfcloutier, 2018-08-31:
No effect

dlech, 2018-08-31:
Do you have battery power?

jfcloutier, 2018-08-31:
Plugged in with power supply

dlech, 2018-08-31:
I'll try to reproduce the problem when I have a chance.

jfcloutier, 2018-08-31:
Is the BrickPi3 board getting power from the Pi3?

dlech, 2018-08-31:
It can be powered from the Pi, but motors don't work in that case.

jfcloutier, 2018-08-31:
I speak at a conference in 10 days and I would love to demo a working robot. Your help is greatly appreciated :-)

jfcloutier, 2018-08-31:
Ah ah! Let me get some batteries!

jfcloutier, 2018-08-31:
That was it. RTFM... https://www.dexterindustries.com/BrickPi/brickpi3-getting-started-step-1-assembly/
Thanks!

closed 2018-08-31 by jfcloutier