ev3dev/ev3dev #1164: C++ or Python? Which language can make my code faster?

lvjonok, 2018-09-02:
I write programs for robotics competition,so i want to now will using of c++ give me more speed in programs?

I ask this question, because now I use Python. I wrote a regulator for controlling motor (MoveEnc) , this function is in class, so i got only 10 iteration in cycle,how I can make my code faster, or I need to use other language.

    def motors(self,vLeft,vRight):
        if vLeft > 100:
            vLeft = 100
        if vLeft < -100:
            vLeft = -100
        if vRight > 100:
            vRight = 100
        if vRight < -100:
            vRight = -100
        self.mLeft.run_direct(duty_cycle_sp = vLeft * self.robot['leftMotor_reverse'])
        self.mRight.run_direct(duty_cycle_sp = vRight * self.robot['rightMotor_reverse'])

    def moveEnc(self,sm,stop = True,speed = 90):
        self.mLeft.position = 0
        self.mRight.position = 0
        encL = abs(self.mLeft.position)
        encR = abs(self.mRight.position)
        min_speed = 50
        max_speed = speed
        speed = min_speed
        startL = encL
        startR = encR
        nB = int(self.sm2cpr(abs(sm)) + (startL + startR)/2)
        while ((encL + encR)>>1 <= nB):
            encL = abs(self.mLeft.position)
            encR = abs(self.mRight.position)
            err = (encL - startL) - (encR - startR)
            p_koef = 2 #1.5
            speedL = speed - err * p_koef
            speedR = speed + err * p_koef
            print(speed)
            speed += 7 if speed < max_speed and (encL + encR)//2 < (nB // 4) else 0
            speed -= 15 if speed > 45 and (encL + encR)//2 > (nB//4) else 0
            self.motors(speedL * self.sign(sm),speedR * self.sign(sm))
            sleep(0.01)
        if stop:
            self.motors(0,0)
            self.stop()

System info (from ev3dev-sysinfo)

Image file:         ev3dev-stretch-ev3-generic-2018-08-06
Kernel version:     4.14.61-ev3dev-2.2.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016535159F2
BOARD_INFO_TYPE=main

WasabiFan, 2018-09-02:
In general, C++ will be faster at runtime than python, but in this case based on the code you posted I imagine the problem is slow code rather than a slow language runtime. Nothing here looks like it would be inherently slow to me (other than the sleep call, of course) but I'd suggest you collect data with [a profiler](https://docs.python.org/3/library/profile.html) to see what is taking time. You might want to remove the `sleep` while profiling so the output is easier to interpret.

lvjonok, 2021-02-09:
Huh, too old question, suppose we should close it

closed 2021-02-09 by lvjonok