ev3dev/ev3dev #1166: Original Pixy connection / communication

donei34, 2018-09-06:
### What are you trying to do?
Use an emulated pixy cam over I2C (actually an openmv running their pixy over i2c emulation)

### What did you expect to happen?
That there would be anything on direct connection between a standard pixy and ev3dev.

### What actually happened?
There was only documentation for the adapter and lego version. Is it just a matter of connecting correctly (which I can easily do) and using the existing driver or is something new required? (i.e think of me as using the original non lego version of the CMUcam5)

dlech, 2018-09-06:
You can use the smbus library for I2C communications. If you search the other issues for "smbus", there are some examples of using it in python with other devices.

You could also use UART instead of I2C.

donei34, 2018-10-02:
Thanks for your help, I have decided to just use UART

closed 2018-10-02 by donei34