ev3dev/ev3dev #1169: PID tuning and empirical value recommendation

wlnirvana, 2018-09-12:
### TL;DR

I am looking for a tested and systematic way to tune the EV3 motor. Empirically working values for the PID parameters would also be appreciated.

### What are you trying to do?

I am a beginner to PID control and was trying to tune an EV3 large motor with the Ziegler–Nichols method. I set `speed_sp=200`, `speed_i=0`, `speed_d=0`. Then I started to increase `speed_p` and collect the `speed` about every 100ms to see if it is stable and smooth.

### What did you expect to happen?

While increasing the `speed_p` parameter, I expected the `speed` to at least reach almost `200` and start oscillating around that.

### What actually happened?

The speed I got was, as a list, `[88, 61, 64, 53, 73, 79, 58, 83, 80, 65, 87, 67, 74, 56, 70, 72, 57, 66, 69, 54, 65, 70, 56, 73, 84, 65, 51, 68, 58, 68, 60, 81, 70, 61, 68, 59, 53, 69, 81, 68, 57, 79, 64, 72, 60, 52, 65, 79, 61, 51, 63, 70, 59, 73, 60, 65, 53, 69, 69, 47, 19]`. It was never close to `200`, even if `speed_p` was already `1000`.

On the other hand, if I set `speed_sp=200` and keep all other parameters the same, the speeds returned would be able to reach 200 but still far away from `500`.

### What hardware and software are you using (including version numbers)?

```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2018-09-12:
The speed measurement at low speeds is greatly improved in ev3dev-stretch. I recommend using ev3dev-stretch instead of ev3dev-jessie.

wlnirvana, 2018-09-13:
Thanks for the update. I'll give the new version a try and come back to update this thread.

By the way, I recall reading somewhere that the maximum `speed_sp` or `speed_p` supported is `1000`. Is it `speed_sp` or `speed_p`?

dlech, 2018-09-13:
`speed_sp`. It actually depends on the motor that is attached. The EV3 medium motor has a higher max value. And it looks like the large motor is actually limited to 1050.

https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/motors/ev3_motor_defs.c#L64