ev3dev/ev3dev #1171: Booting the ev3dev from USB

DeroBee, 2018-09-18:
### What are you trying to do?
I'm trying to setup my ev3 in a way that i can use python to program it. As i had no SD card at hand 
I tried using an USB-Stick, and followed the procedure for flashing.

### What did you expect to happen?
After I plugged the flashed USB into the USB-port next to the mircoSD-port, i expected it to work.
### What actually happened?
It doesnt boot into the ev3dev envoirement instead it boots right into the standards mindstorms envoirement

### What hardware and software are you using (including version numbers)?
The ev3 is running 1.09H. The USB is flashed with ev3dev-stretch-ev3-generic-2018-08-06
The flashing was done with Etcher

### What I'm asking
Is it even possibly to boot from USB or have I made a mistake at some point?

dlech, 2018-09-18:
It seems like I just answered this recently, but I can't find it, so...

Practically speaking, you can't do this.

The bootloader doesn't support USB boot, so you would have to compile your own version of U-Boot (and probably write your own USB driver for it because I don't think there is one for this SoC already). And even if you did all of this, the USB port is USB 1.1 only, so it will be be probably really slow.

DeroBee, 2018-09-18:
Thank you for the answer. 
Seems like i have to find my microSD card.

closed 2018-09-18 by DeroBee