ev3dev/ev3dev #1172: how can i connect pycharm with lego mindstorms...

AprilLeslie, 2018-09-19:
### What are you trying to do?

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

AprilLeslie, 2018-09-19:
I'm doing my project which able to read DXF file and transfer the entities to G-code, then I have to convert coordinates to angle, and set up the lego mindstorms. First, VS code always cant import ev3dev package, and every package.... and the main part of my code is developed using pycharm, I want to connect pycharm with my robot, someone able to tell me where i can find a tutorial for it

dlech, 2018-09-19:
https://www.ev3dev.org/docs/tutorials/setting-up-python-pycharm/

AprilLeslie, 2018-09-20:
I follow the steps, but when I try to push the run_motor.py file, the default is master-> origin: master, after click push button, the log said: Push successful: Pushed 1 commit to origin/master... but i can't find the file in my lego mindstorms ...

AprilLeslie, 2018-09-20:
> https://www.ev3dev.org/docs/tutorials/setting-up-python-pycharm/

I follow the steps, but when I try to push the run_motor.py file, the default is master-> origin: master, after click push button, the log said: Push successful: Pushed 1 commit to origin/master... but i can't find the file in my lego mindstorms ...