ev3dev/ev3dev #1173: 'run-to-rel-pos' behaves like 'run-to-abs-pos' if it was previously run for 0 units

WasabiFan, 2018-09-24:
### What are you trying to do?

Run a motor for 0 units, then for nonzero units, with the `run-to-rel-pos` command.

### What did you expect to happen?

It moves _N_ units, relative to its current position.

### What actually happened?

It treated the setpoint as an absolute position and ran all the way back to position _N_.

### What hardware and software are you using (including version numbers)?
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-07-15
Kernel version:     4.14.55-ev3dev-2.2.0-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165340720B
BOARD_INFO_TYPE=main
```

---

```bash
motor1=/sys/class/tacho-motor/motor0

# Happens with 'coast' too.
echo "hold" > $motor1/stop_action

# Move the motor to a position far away from zero
echo 300 > $motor1/speed_sp
echo 1000 > $motor1/position_sp
echo "run-to-abs-pos" > $motor1/command

read -p "waiting... press enter when motor stops moving"

echo "Running 0 units"
# Ask the motor to go nowhere
echo 0 > $motor1/position_sp
echo "run-to-rel-pos" > $motor1/command

# Delay to give system a chance to process
sleep 2

echo "Position before run:"
cat $motor1/position

echo "Running 100 units"
# Run the motor 100 units forward: rather than move ~1/3 of a rotation, the motor will go all the way back to position 100
echo 100 > $motor1/position_sp
echo "run-to-rel-pos" > $motor1/command

```

This is breaking ev3dev-lang-python's motor pair classes; if you drive with steering 50, one motor is stationary and therefore run for 0 units.

dlech, 2018-09-24:
Does this still happen in the `ev3dev-2.3.0` kernel?

WasabiFan, 2018-09-24:
I've left it upgrading and will test when I get back to my PC.

WasabiFan, 2018-09-24:
@dlech Yes, I still see this with the above script.

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-07-15
Kernel version:     4.14.71-ev3dev-2.3.0-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165340720B
BOARD_INFO_TYPE=main
```

dlech, 2018-11-13:
Fix released in `ev3dev-2.3.1` kernel.

closed 2018-11-13 by dlech