ev3dev/ev3dev #1174: BrickPi3 + RaspberryPi3 how to permanently set up sensors

FRozanski, 2018-09-24:
### System information:

- I am using:
  - RaspberryPi 3 Model B
  - BrickPi 3
  - EV3 motors and sensors(touch, ultrasonic)
- I installed ev3dev using this image file: ev3dev-jessie-rpi2-generic-2017-09-14
- My kernel version is: 4.4.78-21-ev3dev-rpi2
- I am connecting using Ethernet cable
- My changes to config.txt:
```
dtoverlay=brickpi3
dtoverlay=pi3-disable-bt
```

Everytime I turn on my device i must perform following command to set up TouchSensor on port 3
```
echo ev3-analog > /sys/class/lego-port/port3/mode
```

and this two commands to set up UltrasonicSensor on port 2
```
echo ev3-uart > /sys/class/lego-port/port2/mode
echo lego-ev3-us > /sys/class/lego-port/port2/set_device
```
Before applying the above commands:
```
robot@ev3dev:~$ dmesg | grep lego
[   14.384468] lego-port port0: Registered 'spi0.1:S1' on 'spi0.1'.
[   14.384617] lego-port port1: Registered 'spi0.1:S2' on 'spi0.1'.
[   14.384787] lego-port port2: Registered 'spi0.1:S3' on 'spi0.1'.
[   14.384941] lego-port port3: Registered 'spi0.1:S4' on 'spi0.1'.
[   14.385099] lego-port port4: Registered 'spi0.1:MA' on 'spi0.1'.
[   14.385229] lego-port port4: Added new device 'spi0.1:MA:lego-nxt-motor'
[   14.385509] lego-port port5: Registered 'spi0.1:MB' on 'spi0.1'.
[   14.385639] lego-port port5: Added new device 'spi0.1:MB:lego-nxt-motor'
[   14.385804] lego-port port6: Registered 'spi0.1:MC' on 'spi0.1'.
[   14.385929] lego-port port6: Added new device 'spi0.1:MC:lego-nxt-motor'
[   14.386083] lego-port port7: Registered 'spi0.1:MD' on 'spi0.1'.
[   14.386208] lego-port port7: Added new device 'spi0.1:MD:lego-nxt-motor'
[   14.662195] tacho-motor motor0: Registered 'lego-nxt-motor' on 'spi0.1:MA'.
[   14.666511] tacho-motor motor1: Registered 'lego-nxt-motor' on 'spi0.1:MB'.
[   14.669823] tacho-motor motor2: Registered 'lego-nxt-motor' on 'spi0.1:MC'.
[   14.672667] tacho-motor motor3: Registered 'lego-nxt-motor' on 'spi0.1:MD'.
```
After applying the above commands:
```
robot@ev3dev:~/Beta$ dmesg | grep lego
[   14.384468] lego-port port0: Registered 'spi0.1:S1' on 'spi0.1'.
[   14.384617] lego-port port1: Registered 'spi0.1:S2' on 'spi0.1'.
[   14.384787] lego-port port2: Registered 'spi0.1:S3' on 'spi0.1'.
[   14.384941] lego-port port3: Registered 'spi0.1:S4' on 'spi0.1'.
[   14.385099] lego-port port4: Registered 'spi0.1:MA' on 'spi0.1'.
[   14.385229] lego-port port4: Added new device 'spi0.1:MA:lego-nxt-motor'
[   14.385509] lego-port port5: Registered 'spi0.1:MB' on 'spi0.1'.
[   14.385639] lego-port port5: Added new device 'spi0.1:MB:lego-nxt-motor'
[   14.385804] lego-port port6: Registered 'spi0.1:MC' on 'spi0.1'.
[   14.385929] lego-port port6: Added new device 'spi0.1:MC:lego-nxt-motor'
[   14.386083] lego-port port7: Registered 'spi0.1:MD' on 'spi0.1'.
[   14.386208] lego-port port7: Added new device 'spi0.1:MD:lego-nxt-motor'
[   14.662195] tacho-motor motor0: Registered 'lego-nxt-motor' on 'spi0.1:MA'.
[   14.666511] tacho-motor motor1: Registered 'lego-nxt-motor' on 'spi0.1:MB'.
[   14.669823] tacho-motor motor2: Registered 'lego-nxt-motor' on 'spi0.1:MC'.
[   14.672667] tacho-motor motor3: Registered 'lego-nxt-motor' on 'spi0.1:MD'.
[ 2273.055288] lego-port port3: Added new device 'spi0.1:S4:lego-ev3-touch'
[ 2273.062630] lego-sensor sensor0: Registered 'lego-ev3-us' on 'spi0.1:S3'.
[ 2273.066534] lego-port port2: Added new device 'spi0.1:S3:lego-ev3-us'
[ 2273.089390] lego-sensor sensor1: Registered 'lego-ev3-touch' on 'spi0.1:S4'.
```

Is exist any way to set up this ports permanently, instead of doing this every time after turning on device?

dlech, 2018-09-24:
The old fashioned way is to use `/etc/rc.local`.

```bash
sudo sh -c 'cat >> /etc/rc.local << EOF
#!/bin/bash

echo ev3-analog > /sys/class/lego-port/port3/mode
echo ev3-uart > /sys/class/lego-port/port2/mode
echo lego-ev3-us > /sys/class/lego-port/port2/set_device

exit 0
EOF
'
sudo chmod +x /etc/rc.local
```

To make changes later `sudo nano /etc/rc.local`

FRozanski, 2018-09-24:
Thanks, it works!