ev3dev/ev3dev #1175: BrickPi3 + RaspberryPi3 wifi connection

FRozanski, 2018-09-25:
### System information:

- I am using:
  - RaspberryPi 3 Model B
  - BrickPi 3
  - EV3 motors and sensors(touch, ultrasonic)
- I installed ev3dev using this image file: ev3dev-jessie-rpi2-generic-2017-09-14
- My kernel version is: 4.4.78-21-ev3dev-rpi2
- I am connecting using Ethernet cable
- My changes to config.txt:
```
dtoverlay=brickpi3
dtoverlay=pi3-disable-bt
```
I am trying to connect BrickPi via wifi. Im following this [tutorial.](https://www.ev3dev.org/docs/tutorials/setting-up-wifi-using-the-command-line/) 
I have previously connected to wifi network so output is slightly different:
```
robot@ev3dev:~$ connmanctl
connmanctl> enable wifi
Error wifi: Already enabled
connmanctl> scan wifi
Scan completed for wifi
connmanctl> services
*AO Honor 8              wifi_b827eb0ea165_486f6e6f722038_managed_psk
*AR Wired                ethernet_b827eb5bf430_cable
    PLAY-ONLINE-1B1E     wifi_b827eb0ea165_504c41592d4f4e4c494e452d31423145_managed_psk
connmanctl> agent on
Agent registered
connmanctl> connect wifi_b827eb0ea165_486f6e6f722038_managed_psk
Error /net/connman/service/wifi_b827eb0ea165_486f6e6f722038_managed_psk: Already connected
connmanctl> quit
```
Then I do following steps:

- disconnect ssh treminal
- turn off BrickPi 
- plug off ethernet cable
- turn on BrickPi
- and I want to connect via ssh terminal again, but I receive message "Network error: Connection time out"

What I'm doing wrong?

dlech, 2018-09-25:
Did you set `autoconnect on`?

FRozanski, 2018-09-25:
Everything works. I forgot, that wifi connection will have another ip address. Sorry for problem.

dlech, 2018-09-25:
FYI, ev3dev-stretch uses zeroconf/bonjour, so with Windows 10, macOS or Linux, you should be able to use `ev3dev.local` instead of the IP address.

closed 2018-09-25 by dlech