ev3dev/ev3dev #1176: Use Two Remotes Control with BrickPI3

Negociation, 2018-09-29:
### What are you trying to do?
Hi , i posted before here , i trying to do a self driving robot using ev3dev + CNN , now i'm working on dataset algortym generator... 

### What did you expect to happen?
- Because the hardware recently i move from EV3 to Raspberry PI + BrickPI , i'm using Python , now to control my robot i'm using two remote controls , the first on channel 1 to forward and backward , and a second on channel 2 to left and right 

### What actually happened?
My robot only responses to my Control on Channel 1 , why thats happens ? 

My Code:
#!/usr/bin/env python3
from time import sleep
import time
import os.path
from ev3dev.brickpi import *
import threading
import subprocess
import cv2
###################################################################
#		   DECLARACOES DE INICIALIZACAO			   #
###################################################################

#Dados de Revisao
Release = '3'
Version = '1'
Subversion = '2'
Last_State = 'Stop'
StopCapture = False

###########################################################################
#		   	            DATASET  			          #
###########################################################################

class Source:
	def __init__(self):
		#Conjunto de Dados de Objetos Row
		self.dataset = []
		self.target = 'Version = '+ Release + '.' + Version + '.' + Subversion
		self.date = time.strftime("%d%m%Y")
		self.hour = time.strftime("%H:%M:%S")
		self.build = 1
		while os.path.isfile('/home/robot/r'+self.date+'-b'+str(self.build)+'.marcset'):
			self.build += 1
		self.filename = '/home/robot/r'+self.date+'-b'+str(self.build)+'.marcset';
		
	
	def getTime(self):
		return self.date + ' ' + self.hour;
		

	def getBuild(self):
		return self.build
	
	def Insert(self,Row):
		self.dataset.append(Row)
	
	def getSize(self):
		return len(self.dataset)
	
	def getRow(self,Index):
		return self.dataset[Index]

	def getDataset(self):
		return self.dataset

	def getFilename(self):
		return self.filename
	
Dataset = Source()	

[... 281 lines cut ...]
#Bip de Inicializacao
Sound.beep()

def WriteFile():
	MARCset = open(Dataset.getFilename(), 'w+')

	Header = "------------------------------------\n!    Sistema de Captura de Dados   !\n------------------------------------\n! tiagitofranca04@gmail.com        !\n------------------------------------\n"
	SubHeader = '+ Versao da Compilacao < '+str(Release) + '.' + str(Version) + '.' +str(Subversion)+' >\n+ Data da Captura <'+str(Dataset.getTime())+' >\n+ Build < '+ str(Dataset.getBuild()) +' >\n\nMARCSET:\n'
	MARCset.writelines(Header)
	MARCset.writelines(SubHeader)
	for Row in Dataset.getDataset(): 
		Datarow = str(Row.get_RowTempo()) + " " + str(Row.get_RowProximidade()) + " " + str(Row.get_RowLuz()) + " " + str(Row.get_RowClasse()) + "\n"
		MARCset.writelines(Datarow)
	MARCset.close()

while(StopCapture == False):
	LastCount = 0
	sleep(1)
	if (Dataset.getSize() != LastCount):
		LastCount = Dataset.getSize()
		myObject = Dataset.getRow(Dataset.getSize()-1);
		print(str(Dataset.getSize()) + ' >> T: ' + str(myObject.get_RowTempo()) + ' - D: ' + str(myObject.get_RowProximidade()) + ' - L: ' + str(myObject.get_RowLuz()) + ' - C: ' + str(myObject.get_RowClasse()))
	sleep(0.1)

WriteFile()

dlech, 2018-09-29:
You missed an important question:

```
### What hardware and software are you using (including version numbers)?

<!-- Tip: copy and paste the output of `ev3dev-sysinfo -m` here -->
```

dlech, 2018-09-29:
cc: @ev3dev/python

dlech, 2018-09-29:
One thing I noticed is that since you are using BrickPi3 you should be using:

```python
from ev3dev.brickpi3 import *
```

not 

```python
from ev3dev.brickpi import *
```

Negociation, 2018-09-29:
Oh yeah , about that , its because i couldnt find a good documentation or examples of ev3python for BrickPI3 here , i loses 2 days trying to set the ports correctly huahuhauha... i probably tried change the import...

dlech, 2018-09-29:
If this same code worked on the EV3, then I'm guessing this is a kernel driver bug in the BrickPi3 drivers.

https://github.com/ev3dev/lego-linux-drivers/blob/126c76d23443acab0a08b8069960e5308f9f4567/brickpi3/brickpi3_ports_in.c#L199

Looks like `BRICKPI3_SENSOR_TYPE_EV3_INFRARED_REMOTE` from line above should be moved to the group that reads 4 data values in line below.

https://github.com/ev3dev/lego-linux-drivers/blob/126c76d23443acab0a08b8069960e5308f9f4567/brickpi3/brickpi3_ports_in.c#L223

Negociation, 2018-09-29:
Ok , how i do that ? , how can i edit the brickpi3_pors_in.c ?

Negociation, 2018-09-30:
Tanks bro , i'll try now... , I imagine as something about driver , because i read when u started the BrickPI3 Support u basicly had to rewrite the drivers again from scrath , i guess i'm probably the first who trying to use two controls at the same time the reason why i doing that its because as more ergonomic, take a look: 

           LEFT CONTROL                         RIGHT CONTROL
Green: Forward , Red: Stop |  Red Left: Left , Blue Right: Right
[image]

PS:. I tested and not work yet , still only reading the channel 1...

dlech, 2018-09-30:
Run `uname -r` to see if the new kernel is really being used

dlech, 2018-09-30:
And if the picture is recent, one remote looks like it is on channel 4, but the program uses channels 1 and 2.

I like the red/green button mod, by the way. :smile:

Negociation, 2018-09-30:
I changed again to channel 4 on my code before , and i tested on watch values but nothing... huahauha ,

dlech, 2018-09-30:
OK, tomorrow I will get my BrickPi3 out and try again.

Negociation, 2018-09-30:
Tanks for helping me bro... bind away tanks for put the support for OpenCV on EV3DEV image , but its too late for me now huahua i'm using now a smartphone to recorder video...

dlech, 2018-09-30:
[linux-image-4.14.71-ev3dev-2.3.0-rpi2+_1_armhf.zip](https://github.com/ev3dev/ev3dev/files/2431907/linux-image-4.14.71-ev3dev-2.3.0-rpi2%2B_1_armhf.zip)

Please try this kernel. All 4 channels should be working now.

Negociation, 2018-09-30:
Tanks , i was waiting here huahuhauha,

dlech, 2018-09-30:
> so whats happened ?

There were two bugs that needed to be fixed, not just one. You can look at the commit linked above if you want to know the technical details.

Negociation, 2018-09-30:
Next Step for me now  its use some PID with Gyro Sensor...  i dont know why but my robot doesnt run straight

dlech, 2018-11-13:
Fix released in kernel `ev3dev-2.3.1`.