ev3dev/ev3dev #1177: PYTHON SOCKET SERVER BROKEN PIP ON EV3DEV

Negociation, 2018-10-01:
Hi , everyone i need a little help , i'm trying to use  socket TCP to control my Robot using my Tablet... , but i having some issues with broken pip on EV3...

MY ARQUITECTURE:

 BrickPI EV3DEV                      Windows Tablet                   Android Device
     Client               <--->                 Server            <--->            Client

My Server :
`class Servidor:
	sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
	connections = []
	def __init__(self):
		#Dados da Conexao [ IP GERAL | PORTA ]
		self.sock.bind(('0.0.0.0', 8000 ))
		self.sock.listen(10)
		
	def handler(self, c,a):
		while True:
			data = c.recv(1024)
			for connection in self.connections:
				connection.send(data)				
				if not data:
					break;
	def run(self):
		while True:
			c,a = self.sock.accept()
			cThread = threading.Thread(target=self.handler, args=(c,a))
			cThread.start()
			self.connections.append(c)
			print(self.connections)

Meu_Servidor = Servidor()
Meu_Servidor.run()
`

My Client on EV3:
```

#Thread para Servidor ( Cliente )
class Thread_Server (threading.Thread):
	sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
	def __init__(self):
		threading.Thread.__init__(self)
	def sendMsg(self):
		while True:
			msg = 'teste'
			self.sock.send(bytes(msg, 'utf-8'))
			sleep(1)
	
	def run(self):
		iThread = threading.Thread(target=self.sendMsg)
		iThread.start()
		try:
        		self.sock.connect(('192.168.16.110', 8000))
		except:
			print("Nao consegui conectar no servidor , Abortar o Programa")
		while True:
			data = self.sock.recv(1024)
			if not data:
				print ("Disconnected")
				break;
			print(str(data, 'utf-8'))

#Inicializacao da Thread do Servidor
Server = Thread_Server()
Server.start()

```

I tested the same code on my machine using localhost and working fine...

dlech, 2018-10-01:
It looks like you are starting the thread that calls sendMsg before you actually connect the socket.

Negociation, 2018-10-01:
Wow, tanks for open my eyes , well its 2 days programming here , i have to start tommorow my data capture on university , and i need to do a lot o things before , I have to learn how use Kivy to start the video capture of smartphone ... , finish my IronPython Application to start the server...

dlech, 2019-03-25:
Can we close this issue?

Negociation, 2019-03-25:
Of course , i already finished my papper. Anyway tanks a lot for your help and support especially with the issue of IR Controls on brickpi saved my Life :D

dlech, 2019-03-25:
Cool. Can we read your paper? :smile:

closed 2019-03-25 by dlech