ev3dev/ev3dev #1178: Brickman do not return in stretch version of ev3dev

lgneribonfim, 2018-10-07:
I am using the stretch version of ev3dev in SSH connection, because I can't use vs code because of an error.

When a take control of the brickman with command line "sudo chvt 6" and release the control with "sudo chvt 1" I expect brickman returns, but it never happens!!!

dlech, 2018-10-07:
In ev3dev-stretch, use `brickrun` to run your program. It will take care of switching the console for you.

    brickrun -- my-program

dlech, 2018-10-08:
Try:

    brickman -- python3 disp.py

lgneribonfim, 2018-10-10:
Still not working! "brickman -- python3 disp.py" returns "command not found"!
When I try "brickrun -- python3 disp.py" program runs, but with problems!
And the brickman still not returns.
The ev3 screen displays the message below forever.

"        
Starting WPA supplicant...
[  OK  ] Started WPA supplicant.
[  OK  ] Started OpenBSD S▒re Shell server.
[  OK  ] Started Network Name Resolution.
[  OK  ] Started Wait for ▒ured by ConnMan.
[  OK  ] Reached target Network is Online.
         Starting LSB: Start NTP daemon...
[  OK  ] Started LSB: Start NTP daemon.
[  OK  ] Reached target Multi-User System.
         Starting LEGO MIN▒ORMS EV3 LEDs...
[  OK  ] Reached target Gr▒hical Interface.
         Starting Update U▒level Changes...
[  OK  ] Started LEGO MINDSTORMS EV3 LEDs.
[  OK  ] Started Update UT▒unlevel Changes.
[  OK  ] Created slice User Slice of robot.
         Starting User Man▒ for UID 1000...
[  OK  ] Started Session c1 of user robot.
[  OK  ] Started User Manager for UID 1000.
[  OK  ] Started Session c2 of user robot.
[ TIME ] Timed out waiting▒c.1.auto.device.
"

dlech, 2018-10-10:
oops, that was supposed to be `brickrun -- python3 disp.py` as you guessed.

What is the answer to the question that you deleted?

```
### What hardware and software are you using (including version numbers)?

<!-- Tip: copy and paste the output of `ev3dev-sysinfo -m` here -->

```

lgneribonfim, 2018-10-11:
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2018-08-06
Kernel version:     4.14.61-ev3dev-2.2.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165345318D
BOARD_INFO_TYPE=main
```

dlech, 2019-03-25:
Can you post your python program?

KWSmit, 2019-11-21:
I don't know if this is useful information, but I encountered the same problem. But with me it seems to work fine with `brickrun -- python3 my-program.py` as long as you didn't use `sudo chvt 6` or `sudo chvt 1` before. If you did, reboot the EV3 and use the `brickrum`  command. Tested this on Kernel version 4.14.117-ev3dev-2.3.4-ev3.

dlech, 2019-11-21:
or `sudo chvt 2` to get back to brickman without rebooting.

MrMan999, 2020-09-24:
I found a solution—
It was Open Roberta Lab that it tries to load after Brickman.  
To disable Open Roberta Lab, run:
`sudo systemctl stop openrobertalab.service`
`sudo systemctl mask openrobertalab.service`