ev3dev/ev3dev #1182: Connection over bluetooth EV3

ImGek, 2018-10-09:
Hi, I'm trying to connect to my ev3 using bluetooth (I am on Ubuntu Desktop 18.04). 

After a lot of attempts I was able to connect using ssh and send some commands like `fortune`.
However, after 10 seconds I launch the `sudo apt update` command the terminal does not update ETA or percentage and I have to reboot the EV3 and re-establish the connection.

I can figure out what's happening...

dlech, 2018-10-09:
`apt` commands are very slow on EV3. They take several minutes or more. Did you wait that long or only 10 seconds? Also using USB will be faster than Bluetooth.

ImGek, 2018-11-09:
I waited for hours without response, and the console was unresponsive too

staadecker, 2019-04-20:
Might be a duplicate of #1154