ev3dev/ev3dev #1183: I can not connect my mobilephone with my ev3dev ev3

ogwfvjurf, 2018-10-10:
I am trying to controll my ev3 with my mobilephone on my robot is running a python scribt which is working, but when i tried to connect my mobile to the ev3, to be able to send strings as commands there was an error : 
GDBus.Error:org.bluez.Eror.Failed: Protocol not aviable

ev3dev version:
Linux ev3dev 4.14.61-ev3dev-2.2.2-ev3 

do you have an idea how i can solve tis problem or if ther is a better way to controll a ev3dev python with a mobilephone

dlech, 2018-10-10:
To connect a mobile phone via Bluetooth, you need to enable tethering on the mobile phone. Then remove the Bluetooth devices on both the phone and the EV3, then re-pair the devices.

Negociation, 2018-10-17:
U can use a Socket TCP Conection and create a local conection using USB ...

mijoli, 2019-01-07:
I have exatcly the same issue. I am totally newbie with this ev3dev.
Linux version: 4.14.80-ev3dev-2.3.1-ev3
Unable to start bluetooth at all.
journalctl gives more bluetooth related errors like
ev3dev kernel: Bluetooth: hci0 sending frame failed (-49)
I would really want to have one wireless connection.