ev3dev/ev3dev #1184: Program stops, when a cable disconnects

jironer, 2018-10-15:
### What are you trying to do?
Write a program, which doesn't stop when a cable disconnects
I'm sorry for the ironic looking answers, but I think it clears up any confusion, so please don't take this personally. I really need to make it stop stopping whenever a cable disconnects. I want to compete and any non lego parts which would allow me to secure the wires better are forbidden, so I'd like to have a safety, that would ignore error 19 (error which appears, when I disconnect a cable)
### What did you expect to happen?
It wouldn't stop

### What actually happened?
It stopped

### What hardware and software are you using (including version numbers)?
*System info (from `ev3dev-sysinfo`)*
```
Image file:         ev3dev-stretch-ev3-generic-2018-08-06
Kernel version:     4.14.61-ev3dev-2.2.2-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653549EEF
BOARD_INFO_TYPE=main

dlech, 2018-10-15:
Can you post the source code to your program?

jironer, 2018-10-15:
https://pastebin.com/MEBiEZhd
To clarify, I don't know how to do it, and just expected, that it won't do so.
The code has no special features to ignore error code 19, because I'm a total noob to ev3dev.

dlech, 2018-10-15:
You can catch the exceptions. For example replace this:

```python
MotorL.run_direct(duty_cycle_sp=30)
MotorR.run_direct(duty_cycle_sp=70)
```
 with  this:

```python
try:
    MotorL.run_direct(duty_cycle_sp=30)
    MotorR.run_direct(duty_cycle_sp=70)
except OSError:
    pass

This will actually catch all OS errors, not just error 19, but that should be fine.

One side effect of doing it the way I have shown is that if `MotorL` becomes disconnected, then `MotorR` will also stop working because that line will be skipped. When there is an error, the program jumps directly to the `except` line.

jironer, 2018-10-15:
Thank you, this has helped me a lot.