ev3dev/ev3dev #1186: large motor runs too fast with low speed_sp

dwalton76, 2018-10-26:
### What are you trying to do?
I am working on https://github.com/ev3dev/ev3dev-lang-python/issues/392 

I think I have most of the code/math worked out in my MoveDifferential class but I am hitting a snag...when I tell a large motor to rotate at speed_sp 92 for 1408 degrees I expect it to run for 15.3 seconds but it is running much faster than 92 and is finishing in about 13 seconds (so ~108 speed_sp in reality).  There is a lengthy discussion in https://github.com/ev3dev/ev3dev-lang-python/issues/220 where near the end ndward summarizes

`For LOW values of speed_sp, the actual average speed tends to be GREATER THAN the requested speed (speed_sp). The difference can exceed 20%.`

That seems to be what I am seeing.  In that same thread @dlech mentions that it could be an issue with the motor driver and there is reference to issue #543  which sounds like it could have improved this but that was committed in early 2016.

Are there any known outstanding issues with honoring low speed_sp settings for large motors? (I have not tested with medium motors yet).  I looked through the ev3dev issues but did not see any.

### What did you expect to happen?
For the motor to rotate for ~15.3 seconds

### What actually happened?
It finishes in about 13s so is running at speed_sp ~108

### What hardware and software are you using (including version numbers)?

Image file:         ev3dev-stretch-ev3-generic-2017-09-30
Kernel version:     4.9.49-ev3dev-1.5.1-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165346201F
BOARD_INFO_TYPE=main

dlech, 2018-10-26:
Please try with the most recent ev3dev-stretch SD card image.

dwalton76, 2018-10-26:
I should have done that from the start :(  I thought a dist-upgrade would get me to the latest kernel though? (I had done that)

The issue is fixed in
```
Image file:         ev3dev-stretch-ev3-generic-2018-09-24
Kernel version:     4.14.71-ev3dev-2.3.0-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165346201F
BOARD_INFO_TYPE=main
```