ev3dev/ev3dev #1188: brickpi3 motor state is "running holding" once finished instead of just "holding"

dwalton76, 2018-10-31:
### What are you trying to do?

Am testing the fix for #1003 via
```
#!/usr/bin/env python3

from ev3dev2.motor import OUTPUT_A, OUTPUT_B, OUTPUT_C
from ev3dev2.motor import LargeMotor, MediumMotor, SpeedRPM

lmotor = LargeMotor(OUTPUT_A)
lmotor.on_for_seconds(SpeedRPM(40), 10)
```

### What did you expect to happen?

For the motor to rotate for 10s and the program to exit

### What actually happened?

The motor rotates for 10s but foo.py never exits, I ctrl-C and see it is stuck at
```
^CTraceback (most recent call last):
  File "./utils/foo.py", line 7, in <module>
    lmotor.on_for_seconds(SpeedRPM(40), 10)
  File "/home/robot/ev3dev-lang-python/ev3dev2/motor.py", line 975, in on_for_seconds
    self.wait_until_not_moving()
  File "/home/robot/ev3dev-lang-python/ev3dev2/motor.py", line 835, in wait_until_not_moving
    return self.wait(lambda state: self.STATE_RUNNING not in state or self.STATE_STALLED in state, timeout)
  File "/home/robot/ev3dev-lang-python/ev3dev2/motor.py", line 816, in wait
    self._poll.poll(None if timeout is None else timeout)
KeyboardInterrupt
```

The reason being the python library is waiting for ``running`` to no longer be listed in ``state`` (this is the behavior on EV3).  With brickpi3 when the motor finishes moving ``holding`` is added to the ``state`` but ``running`` is never removed
```
robot@pi3[ev3dev-lang-python]# cat /sys/class/tacho-motor/motor0/state
running
robot@pi3[ev3dev-lang-python]# cat /sys/class/tacho-motor/motor0/state
running
robot@pi3[ev3dev-lang-python]# cat /sys/class/tacho-motor/motor0/state
running holding
robot@pi3[ev3dev-lang-python]# cat /sys/class/tacho-motor/motor0/state
running holding
robot@pi3[ev3dev-lang-python]#
```

### What hardware and software are you using (including version numbers)?
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2017-12-13
Kernel version:     4.14.71-ev3dev-2.3.0-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.3
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

dlech, 2018-10-31:
Actually, the bug is with the EV3. `running` means that "power is being sent to the motor", so whenever `holding` is present, `running` should also be present.

dwalton76, 2018-10-31:
Gotcha...will fix this in the python library

dwalton76, 2018-10-31:
Will re-open this in case you want to use it to track making EV3 behave the same way.

dlech, 2018-11-13:
I'm a bit worried about breaking too much stuff I we change the EV3 behavior now. Also, it looks like BrickPi(+) also has the same behavior as the EV3. So only BrickPi3 and PiStorms are different.

So, I think we will "fix" BrickPi3 and see what happens.

dlech, 2018-11-13:
Fix released in kernel `ev3dev-2.3.1`.

dwalton76, 2019-03-24:
hey David, I am still seeing the state as ``running holding``

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2018-12-14
Kernel version:     4.14.94-ev3dev-2.3.2-rpi2
Brickman:           0.10.0
BogoMIPS:           38.40
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=A22082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=000000002f94f0a9
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=E1DA1CC95150324336202020FF19312F
BOARD_INFO_TYPE=aux
```

motor state after running ``./utils/move_motor.py --speed 360 A 720`` which sets the stop_action to ``hold``

```
robot@brickpi3:~$ cat /sys/class/tacho-motor/motor0/state
running holding
robot@brickpi3:~$
```

dlech, 2019-04-08:
Fix released in kernel `ev3dev-2.3.3`.

closed 2019-04-08 by dlech