ev3dev/ev3dev #1189: Reset ev3dev subsystem by command line

jabrena, 2018-11-03:
Hi @dlech,

executing a regression test for the Java library for ev3dev, I observed that the whole /sys/class/ paths disappeared for a strange reason:

``` 
robot@ev3dev:/sys$ ls
block  bus  class  dev  devices  firmware  fs  kernel  module  power
```

How to restart the subsystem in charge of loading ev3dev to map sensor and actuators?

I am using last ev3dev version:

```
ev3dev#15|<!-- Copy everything between these lines -->
ev3dev#15|**System info (from `ev3dev-sysinfo`)**
ev3dev#15|```
ev3dev#15|Image file:         ev3dev-stretch-ev3-generic-2018-08-06
ev3dev#15|Kernel version:     4.14.61-ev3dev-2.2.2-ev3
ev3dev#15|Brickman:           0.10.0
ev3dev#15|BogoMIPS:           226.09
ev3dev#15|Board:              board0
ev3dev#15|BOARD_INFO_HW_REV=7
ev3dev#15|BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
ev3dev#15|BOARD_INFO_ROM_REV=6
ev3dev#15|BOARD_INFO_SERIAL_NUM=001653413CAE
ev3dev#15|BOARD_INFO_TYPE=main
ev3dev#15|```
ev3dev#15|<!-- Copy everything between these lines -->
```

Juan Antonio

dlech, 2018-11-03:
I don't understand the problem. I see that `/sys/class` is still there in your `ls` command.

jabrena, 2018-11-03:
Ohh, you are right, I didn't past the right command.

But basically, I had to restart the brick. For some reason the subystem din't detect a sensor.

I will close this issue. I lost the evidences. :(

Thanks

closed 2018-11-03 by jabrena