ev3dev/ev3dev #1191: ev3dev arch linux connection

B97theFallen, 2018-11-08:
### What are you trying to do?
Connect the EV3 through usb with the tutorial 
( https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/ ).

### What did you expect to happen?
I expected being able to use ssh, like in the ssh tutorial. 
( https://www.ev3dev.org/docs/tutorials/connecting-to-ev3dev-with-ssh/ )

### What actually happened?
I reached a state of being "connected", but not "online" as the tutorial suggested.
The IP on EV3 screen: 169.254.x.x (where x.x change each resconnect)
As a result I get "ssh: Could not resolve hostname ev3dev.local: Name or service not known" from "ssh robot@ev3dev.local".

### What hardware and software are you using (including version numbers)?
I'm using arch linux ( kernel: 4.18.16-arch1-1-ARCH )
I used the "nm-connection-editor" (launched through terminal) to perform most of the tutorial.
Brickman v0.10.0 ( kernel: 4.14.61-ev3dev-2.2.2-ev3 )
I'm connecting through the included usb
I'm using ethernet on my host PC.

First time working with EV3 and ev3dev, so I might have missed something obvious to you, I'll appreciate any feedback.

dlech, 2018-11-08:
I'm guessing that NetworkManager on Arch must have a different default configuration or something that is preventing it from working. You might have better luck asking about this on Arch forums.

As for the host name resolution, you need to have Avahi installed for that to work.

TpouHuK, 2019-02-05:
1. Connect EV3 via USB, but don't be connected, if connected then disconnect.
2. Launch `nm-connection-editor`, edit connection to shared.
3. On EV3 check in IPv4 default Linux settings and press connect.

SSH via IP:
`$ ssh robot@10.42.0.3`

Issues:
All these steps should be reproduced after every USB disconnection, as MAC in connection changes.

dlech, 2019-02-08:
> All these steps should be reproduced after every USB disconnection, as MAC in connection changes.

Please open a new issue for this and make sure to include all of the information requested from `ev3dev-sysinfo`.